Klemen1337/node-thermal-printer #1: Image printing

smanaqvi83, 2015-09-10:
Can you please provide help on printing of images in this printer? using the same API? 

Other printing is working fine, but we need to print images. Please help. 

Regards
Ali

Klemen1337, 2015-09-14:
Hi Ali,
currently we are too busy to update the library. Here are some user manuals for star printers if that helps you. There is a good description how to print dot matrix graphics.
http://www.i-pos.nl/download/star/Prog%20manual%20TSP700.pdf
http://www.star-m.jp/eng/service/usermanual/linemode_cm_en.pdf

We hope to release a new version with image and qr printing in the next week or two.

Cheers,
Klemen

smanaqvi83, 2015-09-14:
Hi,
Thanks. Waiting desperately for this update.

Regards
Ali

zettam, 2015-10-03:
Any updates about this? Much appreciated!

jmercha, 2016-02-17:
I have [fiddled around a little](https://github.com/jmercha/hue-ambience/blob/master/lib/color-mapper.js) with bitmaps in node before.

I also need image printing support for my projects as well, so I'd love to help out here! :smile

Klemen1337, 2016-04-15:
@jmercha I see you made image.js to support image printing. Any luck?

jmercha, 2016-04-16:
@Klemen1337 I got stuck trying to send the right escape codes to the printer and haven't looked at it since. I'll probably have another look next weekend.

Klemen1337, 2016-04-18:
@jmercha, I just changed the library to work with buffer instead of strings. It should help with the problem you have.

More about buffer:
[https://nodejs.org/api/buffer.html](https://nodejs.org/api/buffer.html)

sadlyblue, 2016-06-11:
Hi, I'm trying to insert an image for an Epson thermal printer, with no succes so far.
Can anyone help?
I looked at the epson references, but they are a little confusing to me.
Thanks

Klemen1337, 2016-06-11:
@sadlyblue sorry this library dose not support image printing yet 😢

sadlyblue, 2016-06-11:
I'm aware it doesn't support images yet. But could you help me configure it. If i'm able to to it, i'm happy to provide the code for integration in your library.
The main difficulty i have, after looking at the epson references is how should i calculate the pL and pH values (along with all the other values), in this function GS ( L pL pH m fn a bx by c xL xH yL yH d1...dk, and i guess i should pass the image as hex as well to the buffer.
After, i think i should use the https://reference.epson-biz.com/modules/ref_escpos/index.php?content_id=98 function 50 to print the image.
I'm trying to insert a fixed logo.

pratikkulshreshth, 2016-08-04:
Any progress on this?

dustinbolton, 2016-08-05:
Here's a library that can print images to a thermal printer BUT it is not an Epson or Star.  The command codes look different so I don't know if the image format itself is compatible but it may be helpful to someone who is further along in understanding the image formats this expects...

Library:
https://github.com/xseignard/thermalPrinter/blob/master/src/printer.js

Function:
 printImage() and printImageData() in this libr

Klemen1337, 2016-09-07:
I just pushed an experimental :warning:  epson and star image printing, anyone knows why star printing doesnt work? If anyone can help that would be great :smile:

Klemen1337, 2016-09-09:
UPDATE: @SloInfinity just pushed an update for EPSON image print. This library is now supporting PNG image printing for EPSON printers. He is still working on STAR support.

Klemen1337, 2016-09-12:
Image printing done :smile:

closed 2016-09-12 by Klemen1337