Klemen1337/node-thermal-printer #2: network init

c3959, 2015-09-14:
You can add as feature connect via network?
not only by printing through usb

Klemen1337, 2015-09-15:
Hi @c3959,
we do not have network printers, so we didn't include that function. When we get our hands on the network printer we will support it.

This is a forked repository where they support net communication. 
https://github.com/Americas/node-thermal-printer/.

Cheers,
Klemen

eatsleeplim, 2015-10-10:
is it possible have a function that returns (and one that clears) the printText? This would allow users (me!) to print using other means (like the npm printer module) without being tied down to using /dev/usb/lp0.

Klemen1337, 2015-10-21:
@eatsleeplim I have added the clear and getText functions you have requested. We are still working on different kinds of communication and image printing. Currently we only have the USB printer, but we are trying to get the ethernet or serial thermal printers.

Klemen1337, 2015-11-24:
@c3959 can you test the new version with network printing? I dont have a network printer :stuck_out_tongue_winking_eye:

c3959, 2015-11-27:
this weekend will use a printer from the office for testing. I will discuss them how everything works. :D Cheers

kikolobo, 2018-02-04:
Did it worked ?? Can you connect to several printers?

Klemen1337, 2019-02-26:
Network printing is supported

closed 2019-02-26 by Klemen1337