Klemen1337/node-thermal-printer #4: 2 printers connected

dantero00, 2015-11-04:
I have a problem.. i need to connect two thermal in the same computer.. do you have an example about it ?

eatsleeplim, 2015-11-11:
in the current code, the printer is hardcoded to write to `/dev/usb/lp0`.

I'm doing a walkaround by getting the print output and then sending this output through the npm `printer` module (i'm printing using CUPS. My printer is a networked printer).

```
var printer = require('printer');
var doc = require('node-thermal-printer');

var printer_name = 'STAR700'; // the name of my printer in CUPS

function add_printjob(table, menu_items){
  var output = prepare_printout(table, menu_items);
  printer.printDirect({
    data: output,
    type: "RAW",
    printer: printer_name,
    success: success_handler,
    error: error_handler
  });
}

function prepare_printout(table, menu_items){
  doc.init('star');
  doc.println('hello');
  ...
  var output = doc.getText();
  doc.clear();
  return output;
}

// do the printing
add_printjob(1, [{quantity:1, name: "cheesecake"}]);
```

Klemen1337, 2015-11-24:
Hi all,
I added new init configuration where you must set the correct interface directory, so you can add more printers. 
@eatsleeplim your solution is good and it can work with different operating systems.

closed 2015-12-03 by Klemen1337