Klemen1337/node-thermal-printer #7: How to provide printer port in Windows

luckyagarwal3247, 2016-04-22:
Hi,

i want to use this in a windows box. What should be the interface value for the printer connected to port USB001 ?

Thanks
Lucky

Klemen1337, 2016-05-12:
Hi @luckyagarwal3247,
you can use https://www.npmjs.com/package/printer package. just send getBuffer() as data.

closed 2016-05-12 by Klemen1337

luckyagarwal3247, 2016-05-12:
Correct me if I'm wrong,

I am supposed to use your module to generate the ESCP codes and supply the getBuffer() output to the above modules print command as data.

Thanks for the response.

Klemen1337, 2016-05-13:
@luckyagarwal3247 Thtat is correct

jackyzonewen, 2016-06-08:
@Klemen1337 Thanks for your response, i use your module to generate ESCP Command Code, in example.js file, use code "interface: '/dev/usb/lp0'" to access printer, i have an printer with usb interface, how can i send the buffer data to the printer ?, i try to use node-usb, but the printer no response for me, is my method correct？

Klemen1337, 2016-06-11:
@jackyzonewen Im not familiar with node-usb library, by just looking at it it seems that .transfer(data, callback) should work.

seiyria, 2016-12-30:
So based on this, is it safe to say that this library is basically linux only? I can't get it working on OSX (I can't figure out what the interface should be) and I'd need to use it on windows at bare minimum for the built application.

Klemen1337, 2017-01-05:
@seiyria well I'm using it with [printer](https://www.npmjs.com/package/printer) package on windows. But yes the library on its own only works on linux. Maybe windows support will be added in the future.

asoneji, 2017-02-01:
How can we print on mac?

I got the https://www.npmjs.com/package/printer pkg and did https://github.com/tojocky/node-printer/blob/master/examples/getPrinters.js and I get the following output for my thermal printer.

`  { name: 'Star_TSP143__STR_T_001_',
    isDefault: false,
    options:
     { copies: '1',
       'device-uri': 'usb://Star/TSP143%20(STR_T-001)?location=40110000',
       finishings: '3',
       'job-cancel-after': '10800',
       'job-hold-until': 'no-hold',
       'job-priority': '50',
       'job-sheets': 'none,none',
       'marker-change-time': 1970-01-01T00:00:00.000Z,
       'number-up': '1',
       'printer-commands': 'none',
       'printer-info': 'StarTSP143',
       'printer-is-accepting-jobs': 'true',
       'printer-is-shared': 'false',
       'printer-location': 'MTVL1542fcc6f',
       'printer-make-and-model': 'Star TSP100 Cutter',
       'printer-state': '3',
       'printer-state-change-time': 2017-02-01T01:06:49.000Z,
       'printer-state-reasons': 'none',
       'printer-type': '2150404',
       'printer-uri-supported': 'ipp://localhost/printers/Star_TSP143__STR_T_001_' },
    status: 'IDLE' },
`
And tried the following with bellow interface:
interface: 'usb://Star/TSP143%20(STR_T-001)?location=40110000'
interface: 'ipp://localhost/printers/Star_TSP143__STR_T_001_'

`var printer = require("node-thermal-printer");
printer.init({
  type: 'epson',
  interface: '/dev/usb/lp0'
});
printer.alignCenter();
printer.println("Hello world");
printer.printImage('./assets/olaii-logo-black.png', function(done){
  printer.cut();
  printer.execute(function(err){
    if (err) {
      console.error("Print failed", err);
    } else {
     console.log("Print done");
    }
  });
});`

Still not able to print anything. I am on a macOS.

kkaptan, 2017-03-17:
@asoneji, as @Klemen1337 described using this module on Windows/OSX you need to make use of the printer module for interfacing with the printers. And use the node-thermal-printer only to generate the data to be sent to the printer via the `printer.printDirect({data:..,}) method`. 

In your case,
```js
var thermal_printer = require("node-thermal-printer");
var printer = require("printer");

var star = printer.getPrinter('Star_TSP143__STR_T_001_');

thermal_printer.init({
    type:'star'
})

thermal_printer.println("Hello World");
thermal_printer.drawLine();

console.log(thermal_printer.getBuffer());

printer.printDirect({
    data: thermal_printer.getBuffer(),
    printer: star.name,
    type: "TEXT",
    success: function(job_id){
        console.log('OK :'+job_id);
    },
    error: function(err){
        console.error(err);
    }
});
```

Only Problem I'm having is printing barcodes or qrcodes, cut paper for example. Only type that is working for me is `TEXT`.  @Klemen1337 any idea on how I might be able to do this? Maybe you have encountered something similar.

Thermal printer Im using is the TSP100 FuturePRNT, MacOS 10.11.6, node-therminal-printer with printer module.

Klemen1337, 2017-03-23:
Hi @kkaptan, type `TEXT` is not valid. The only avalible types are `star` and `epson`. If you check [line 17](https://github.com/Klemen1337/node-thermal-printer/blob/master/node-thermal-printer.js#L17) you will notice that if the type is not valid `epson` configuration will be used. I know, its bad :smile: 

I believe that TSP100 is a graphical only printer. You have to set it up so it behaves as a 'normal' printer in the software that is included.

Cheers :beer: ,
K

E-P1983, 2017-04-28:
@kkaptan did you ever find out how to print barcodes? :) having the same issue right now

kkaptan, 2017-04-30:
@E-P1983 nope, I stopped trying. I don't think it is supported by the thermal_printer library for the `star` printer.  I believe @Klemen1337 misunderstood my question as I did indeed use `star` for the type as shown in the example provided. I use `type`: `TEXT`, to the `printer` library. 

As a side note, I managed to print an image using just the [printer](https://github.com/tojocky/node-printer) module. But still this was not what I wanted.

SanthoshDhandapani, 2017-06-30:
After a lot of research, atlast i found a working solution.
For Windows:

Install corresponding printer driver
Enable share printer and give a share name from printer properties by right clicking on the printer
Now input the interface in printer config as shown below
Syntax: \\IP_ADDRESS\SHARED_PRINTER_NAME
Example: \\192.168.2.20\Posiflex

SanthoshDhandapani, 2018-04-13:
Hi @IamSumitJangid
Did you tried the above solution which I posted, that worked for me in Windows. I had the same problem and at last that worked for me.

SanthoshDhandapani, 2018-04-14:
@IamSumitJangid Comment that url and also mention your printer name and model

IamSumitJangid, 2018-04-14:
@SanthoshDhandapani 

URL : https://www.youtube.com/watch?v=MQMD4Pb7jzY
Printer name : Tvs RP3160 star.

SanthoshDhandapani, 2018-04-14:
@IamSumitJangid it's clear from that video, they are running in a local environment not from the hosted static webpage. If you clearly observe you can see the localhost in the url. They are using asp as server side programming so thay they are interacting with the system resources and we can achieve the same with node also. 

Clearly think about this,  if the web page is hosted in production which means it can be accessible in any user's browsers throughout the web so you cannot access their file system's, printers and other resources from a website.

SanthoshDhandapani, 2018-04-14:
@IamSumitJangid But there are some ways to communicate to a printer from websites. They are

1. Print by email
HP printers support ePrint, which is the ability to email a doc, pdf, html, or jpg to a printer. It will print automatically, without going through a PC

2. Print by Cloudprint
Another choice, especially if you want to check if the printer is online or if the print job completed, is to use Google Cloudprint. I haven't implemented anything with Cloudprint, but it has a RESTful API:https://developers.google.com/cloud-print/docs/appDevGuide. On the user's side, they can use a printer that directly supports Cloudprint. 

3. Print by local network
If you know the printer is on the local network, such as when you've got a centralized reporting server for a company, then you've got a third option. Assuming your app server is on a Linux box, you can use CUPS to send a job to the printer. Then you can use lpr to send a PDF file to the printer. (Similar to that youtube video) 

So we can access a cloud printer from websites, unfortunately thermal printers cannot be cloud printers. Your printer Tvs RP3160 star. Is also a thermal printer. 

Once you understand these try to explain this to your client or management and mention that without a cloud printer we can print using local environment but not from the production web site.

IamSumitJangid, 2018-04-14:
@SanthoshDhandapani 
thank you brother, I will try above described message and  i'll contact you again if i stuck.

IamSumitJangid, 2018-04-14:
@SanthoshDhandapani 
I got a solution i created a program in java which prints the content within 1 second.Via setting the content type html/css, I passed the hard coded html file which prints the same. Which than integrated in API and called automatically whenever any invoice is generated, Now the problem is how that program will get data dynamically to print. Can i send data via JSON file or via APIcall or can i use angularjs to make API call and render html dynamically.

IamSumitJangid, 2018-04-14:
@SanthoshDhandapani 
Yes my web app is on local system.
Is there  any method to pass data from browser to the created java application for print.

SanthoshDhandapani, 2018-04-14:
Thorough rest api. I hope you can do the remaining stuffs then :)

KakarN, 2018-10-13:
@kkaptan I was trying to `cut()` and `partialCut()` using the same method as yours, but, neither of those commands worked. Did it work for you? If it helps, I am using EPSON TM_T82 S_A, via USB.
```
var thermal_printer = require("node-thermal-printer");
var printer = require("printer");

var epson = printer.getPrinter('EPSON_TM_T82_S_A');

thermal_printer.init({
	type: 'epson'
})

thermal_printer.print("Hello World");
thermal_printer.println("Hello World");
thermal_printer.println("Before partial cut...");
thermal_printer.partialCut();
thermal_printer.println("After partial cut...");
thermal_printer.cut();
thermal_printer.print("Hello World");
thermal_printer.println("Hello World");
thermal_printer.println("Before partial cut...");
thermal_printer.partialCut();
thermal_printer.println("After partial cut...");
thermal_printer.drawLine();

console.log(thermal_printer.getBuffer());

printer.printDirect({
	data: thermal_printer.getBuffer(),
	printer: epson.name,
	type: "TEXT",
	success: function (job_id) {
		console.log('OK :' + job_id);
	},
	error: function (err) {
		console.error(err);
	}
});```

KakarN, 2018-10-14:
@kkaptan Yep! That was it. Thanks! Although, I am unable to print an image. How did you figure out to print the image? I would really appricitate your help. Thank you.

This is the snippet:

```
thermal_printer.println("Hello world");
thermal_printer.printImage('./assets/main-logo.png', function (done) {
	thermal_printer.execute(function (err) {
		if (err) {
			console.error("Print failed", err);
		} else {
			console.log("Print done");
		}
	});
});

printer.printDirect({
	data: thermal_printer.getBuffer(),
	printer: epson.name,
	type: "RAW",
	success: function (job_id) {
		console.log('OK :' + job_id);
	},
	error: function (err) {
		console.error(err);
	}
});
```

But it does not print an image, instead I am getting an error:

```
kakar@Apples-MacBook-Pro electron-pos $ node image.js
OK :80
fs.js:653
  binding.open(pathModule.toNamespacedPath(path),
          ^

TypeError: path must be a string or Buffer
    at Object.fs.open (fs.js:653:11)
    at Object.fs.writeFile (fs.js:1300:6)
    at self.running (/Users/kakar/electron-apps/electron-pos/node_modules/write-file-queue/index.js:53:12)
    at Immediate.<anonymous> (/Users/kakar/electron-apps/electron-pos/node_modules/dank-do-while/index.js:27:7)
    at runCallback (timers.js:756:18)
    at tryOnImmediate (timers.js:717:5)
    at processImmediate [as _immediateCallback] (timers.js:697:5)
```

afranioluizsouzacoelho, 2020-01-10:
Hy, I´m from Brasil and i getting a lot of difficult to print on node-thermal-printer. I´ve tried all od thinks above and i get the message **TypeError: fs.existsSync is not a function**.

I´m work in windos 10.

and the definition are:

var thermal_printer = require("node-thermal-printer");
            var printer = require("printer");

            var epson = printer.getPrinter('EPSON_TM_T82_S_A');

            thermal_printer.init({
                type: 'epson'
            })

            thermal_printer.print("Hello World");
            thermal_printer.println("Hello World");
            thermal_printer.println("Before partial cut...");
            thermal_printer.partialCut();
            thermal_printer.println("After partial cut...");
            thermal_printer.cut();
            thermal_printer.print("Hello World");
            thermal_printer.println("Hello World");
            thermal_printer.println("Before partial cut...");
            thermal_printer.partialCut();
            thermal_printer.println("After partial cut...");
            thermal_printer.drawLine();

            console.log(thermal_printer.getBuffer());

            printer.printDirect({
                data: thermal_printer.getBuffer(),
                printer: epson.name,
                type: "TEXT",
                success: function (job_id) {
                    console.log('OK :' + job_id);
                },
                error: function (err) {
                    console.error(err);
                }
            });

May someone help me? thanks

I'believe that my fs package is not for windows. if so where can i get one for windows?

afranioluizsouzacoelho, 2020-01-10:
More information:

TypeError: fs.existsSync is not a function
(anonymous function)
C:/ProjetosVSCode/drugstorereact/node_modules/printer/lib/printer.js:9
   6 |     binding_path = path.resolve(__dirname, './node_printer.node'),
   7 |     printer_helper;
   8 | 
>  9 | if (fs.existsSync(binding_path)) {
  10 |   printer_helper = require(binding_path);
  11 | } else {
  12 |   printer_helper = require('./node_printer_' + process.platform + '_' + process.arch + '.node');

RahulMongia, 2021-02-02:
Anyone knows how to set-up a USB printer for node js. I m not able to configure the interface for it . Here is my code

enter code here const ThermalPrinter = require("../node-thermal-printer").printer;
                const Types = require("../node-thermal-printer").types;
                const electron = typeof process !== 'undefined' && process.versions && 
                                  !!process.versions.electron;

                async function testConnection () { 

                let printer = new ThermalPrinter({
                type: 'star',                                    
                interface: "Gold" //name of the printer
                });
                printer.print("hi");
                let isConnected = await printer.isPrinterConnected() 
                console.log("Printer connected:", isConnected);
                }
                 testConnection();