Klemen1337/node-thermal-printer #8: Special Characters not displaying properly

ChazUK, 2016-05-31:


Klemen1337, 2016-06-11:
@ChazUK I will try to add support as soon as I can.

sadlyblue, 2016-10-04:
I don't really know how to work with git and github, but i've made a fork to include support for Windows 1252 charset to the epson. If you are interested, just add to the main development.

ntyrrell, 2017-07-27:
Hi @sadlyblue, could you upload your fix as a code block for forking the repository?

sadlyblue, 2017-07-31:
Hi @ntyrrell, sorry for the late reply. I'm not very familiar how to use github. But i've added the code to a new fork. I think you can find it in my account. Basically I added the charSet WPC1252, and the corresponding special characters.

ntyrrell, 2017-07-31:
Hi @sadlyblue, no worries! I managed to spot that, thank you for sharing 😄 In the end, I ended up doing my own fork as I am using a Star printer, but my hope is that it allows for specifying custom special characters (see #32)

closed 2019-12-13 by Klemen1337