Klemen1337/node-thermal-printer #17: Polling printer for status? (eg if out of paper)

dustinbolton, 2016-08-09:
Do you know of any way to be able to get data back from the printer? Specifically I'd like to check the sensors on whether the printer is out of paper.

It looks like there's not currently the ability to read from the printer, only send to it.  Do you think reading in similar to the following might work? I'll try and see what I can figure out in the next few days.

http://stackoverflow.com/questions/2731624/how-to-get-status-from-pos-printer

Klemen1337, 2016-09-02:
@dustinbolton I haven't tried reading data from printer yet. Have you figure anything out yet?

dustinbolton, 2016-09-28:
@Klemen1337 I've been busy the past few weeks but good news: I've been able to grab the status flags from the printer so I'll work on adding that in.

dustinbolton, 2016-11-03:
@Klemen1337 Getting close on this. It turned out to be a lot more intense than I expected getting it to all play nicely together but it should be really cool. Adding support for polling the printer for its status as well as using the "auto status back" feature of these printers allowing it to trigger a callback live any time the status changes.

Klemen1337, 2016-11-14:
What printer were you working on?

closed 2016-11-14 by Klemen1337

dustinbolton, 2016-11-14:
@Klemen1337 I'm working on adding it for both Epson & Star compatible printers. I've tested it on Epson & Citizen brand printers but have not had time to try it out on Star yet although I've added all the command codes so it may already be working for Star. I had to touch quite of bit of code to do all this.

I've pushed my preliminary version up here:
https://github.com/dustinbolton/node-thermal-printer

There are two main new public functions (shown in the documentation): getStatus() and asb(). Now upon init() it opens a connection to the printer and holds it open, passing around a file descriptor for read and write access.

Overview of changes made so far:
Added getInterface()
Added command codes for ABS toggling for Epson & Star
Added command codes for getting current status overview, offline status details, error details, and paper status for Epson & Star
Added bidirectional ability allowing reading of information from printer such as status (online/offline, ready to print, out of paper, jam, etc)
Added function for turning ABS on or off with callbacks on status changes
Added functions for getting various status information
Added configuration codes for Epson & Star printers for status retrieval and auto status back enabling/disabling
Added optional callback to init() to catch errors opening printer.
init() now opens connection to printer immediately and holds open.
Added disconnect() function to disconnect from printer. Undoes init().
Modified write-file-queue package to support file descriptors.
Renamed forked write-file-queue to write-file-queue-fd to prevent accidental overwrite as original package is no longer compatible. Appears fairly abandoned so should be fine.
On execute() and raw() if fd not set then wait 1sec and try again.

NOTE: asb() and getStatus() not yet available for IP printers.

AVanVlack, 2017-07-30:
@dustinbolton  are you still up for merging your code? I have this working on lan, usb (cross platform via libusb), and serial and wanted to build this into the library.