Klemen1337/node-thermal-printer #26: I get stocked in printer.execute(...)

ivanfelipecp, 2017-02-21:
Hello, i get stocked in printer.execute()
this is my code

var printer = require("node-thermal-printer");

printer.init({
  type: 'epson',
  interface: '/dev/usb/lp0'
});

printer.println("Helloworld");
printer.execute(function(err){
    console.log("enters");
    if (err) {
      console.error("Print failed", err);
    } else {
     console.log("Print done");
    }
    printer.clear();
});

The program dont execute the line "console.log("enters");"

KostadinP, 2017-03-02:
Try to run your script as admin. I had the same problem. Is there any way to avoid it?

Klemen1337, 2017-03-02:
Well no because /dev/usb/lp0 is usually non writable for the normal user.

closed 2018-04-17 by Klemen1337