Klemen1337/node-thermal-printer #30: Compile the package down to ES5

rambabusaravanan, 2017-06-27:
**Problem**
Uglify doesn't understand the ES6 syntax.
`let i = "QRCODE_CELLSIZE_".concat(settings.cellSize.toString())`

**Explanation**
I'm building a cross-platform desktop [app](https://github.com/rambabusaravanan/error-create-react-app) using electron and react and I'm using npm package 'node-thermal-printer' for usb thermal printer. The printer code works fine in the places where react-scripts build doesn't involve like
- static .js (public folder) and linked to index.html (renderer process)
- electron's main process (main.js)

But the build fails, when I add the same code in src/index.js where react-scripts build failed with error code ELIFECYCLE, errno 1

**Issue Reference**
https://github.com/facebookincubator/create-react-app/issues/2640

**Expected Solution**
The library needs to be compiled down to ES5 before publishing

**Suggested Work Around**
It's mandate for user to use babel for transpilation at their end.

AVanVlack, 2017-07-05:
Ya sorry this is my code. Was not following the projects style. Fixed here: [](https://github.com/Klemen1337/node-thermal-printer/pull/31/commits/f4b6f513452b88160bc084ec36072cb1a5064701)

rambabusaravanan, 2017-07-19:
Thanks for your contribution. It would be helpful for us if published in npm too.

Klemen1337, 2017-07-20:
@rambabusaravanan Updated to version 1.0.5 on npm, sorry for the delay.

closed 2017-07-20 by Klemen1337