Klemen1337/node-thermal-printer #33: Can't print images/barcodes/QR codes

megaed, 2017-08-13:
I have successfully printed texts and was able to confirm that most of the text options work but...

I noticed that I can't print anything else. I tried transferring an image onto my raspberry pi which is the computer utilizing the thermal printer, it was in PNG format. I then followed the examples and it just won't print out the image. It did move the thermal paper but nothing was printed.

When I tried printing out barcodes -- same result.. barcodes. I tried printing different formats.

I then moved onto the QR code hoping that would work but... no luck.

AVanVlack, 2017-08-13:
What printer are you using?

megaed, 2017-08-13:
https://www.banggood.com/POS-5890K-58mm-Thermal-Receipt-Printer-Support-WIndows-Linux-p-1053549.html

I'm using this printer.

AVanVlack, 2017-08-13:
It looks like that printer supports the basic esc/pos printer commands but is not able to build QR codes in the printer itself. You will have to uses another library to build a raster image of the QR code.

The 1d barcodes should work but is dose not seem to support the hri font command and the way I have it implimented right now it always sends all the setting commands and most likely confuses the printer. Will rethink that implementation but if you go into the library and delete out that line it might work. Lines 461-467 in node-thermal-printer.js

The image printing command in this library is using a older  command to print the images that your print seems to support. Not sure how much it would take to convert it to use the newer commands.

megaed, 2017-08-13:
I applied the fix you suggested for the barcode and it started to work.

Since I can't build the QR code within the printer itself and I have to be able to generate a raster image of the QR code that means -- I still have to print the QR code as an image which I still can't seem to get working.

You said that the image printing should work on my printer but no image is sent to my printer after calling the printImage function...

Is there a way to get this working? Thanks!

5zirok, 2018-01-22:
I also use such a 5890K printer. And I can not print the Cyrillic alphabet. What can be wrong?

thamjieying, 2018-09-06:
I can't print image as well, does anyone know what is the problem?

closed 2019-02-28 by stale[bot]