Klemen1337/node-thermal-printer #44: Windows Printer Star

JefferyHus, 2017-12-11:
```javascript
printer.init({
    type: 'star',
    interface: 'tcp://192.168.1.10:9100',
    characterSet: 'JAPANESE',
    removeSpecialCharacters: false,
    replaceSpecialCharacters: true,
    extraSpecialCharacters:{'円':163}
})

printer.isPrinterConnected( function(isConnected){
    console.log('Printer is connected: ', isConnected)
} )

printer.alignCenter()

printer.print("日本は素晴らしい")
printer.println("Test")
printer.println("Qrcode 128")
printer.code128("granpark")
printer.cut()
printer.execute(function (e) {
    if (e)
    {
        console.log("Print Failed:", e)
    }
    else
    {
        console.log("Printed")
    }
})
```
This is my code, I would love to know what are the interfaces and how we set one, because I believe I am missing something, However the console does print "Printer is connected: true" but still it isn't printing.

AVanVlack, 2017-12-11:
That looks correct. Interfaces get selected automaticly by parsing the location you provided. It looks like you have a network printer? What is the printer model number?

JefferyHus, 2017-12-11:
Yes a network printer, Star TSP143LAN Cutter

AVanVlack, 2017-12-11:
That printer dose not support esc/pos commands for generating receipts. The future print models expect a image of what you want to print that use separate commands then what this library uses. This library won't do much for that printer.

The windows star software dose have an emulator that will take the commands this library generates and produces and image for the printer but if really meant for using the printer with lagacy software.

I am working on a library based on this library that will support this printer but it's currently a mess right now.

JefferyHus, 2017-12-11:
Hi,
I got it to work, it's 6:30 Am here didn't sleep till I fixed this, so forgive my brain dead ideas.
I have been looking inside the Printer library's issues [Printer](https://github.com/tojocky/node-printer/issues/159), thanks to this I have merged both libraries and got it to work, However i'm still struggling at executing the cut, beep commands even tho I feed the Buffer.

```javascript
const printer = require("printer")
const printerTherm = require("node-thermal-printer")
const util = require('util')

printerTherm.init({
    type: 'star',
    interface: 'tcp://192.168.1.10:9100',
    characterSet: 'JAPANESE',
    removeSpecialCharacters: false,
    replaceSpecialCharacters: true,
    extraSpecialCharacters:{'円':163}
})

printerTherm.isPrinterConnected( function(isConnected){
    console.log('Printer is connected: ', isConnected)
} )

printerTherm.alignCenter()

printerTherm.print("日本は素晴らしい")
printerTherm.println("Test")
printerTherm.println("Qrcode 128")
printerTherm.code128("granpark")
printerTherm.cut()
printerTherm.execute(function (e) {
    if (e)
    {
        console.log("Print Failed:", e)
    }
    else
    {
        console.log("Printed")
    }
})

printer.printDirect({
    data: printerTherm.getBuffer(),
    type: 'RAW',
    success: function (id)
    {
        console.log("Printer ID: ", id)
    },
    error: function (err)
    {
        console.log(err)
    }
})
```

I can't think at the moment as I should sleep, thanks for your fast reply :+1:

Klemen1337, 2017-12-11:
Hi @JefferyHus,
No need to printerTherm.execute if you are using printer.printDirect, as you are only using printerTherm for the commands. TSP143LAN might have different codes to execute cut or beep, that is why it is not working.

JefferyHus, 2017-12-11:
Thanks @Klemen1337 Yeah I will try to read its manual and integrate later to work, thanks :+1:

JefferyHus, 2017-12-12:
sorry for being here again, but the Japanese is not well printed

JefferyHus, 2017-12-12:
this is the [Manuel](http://www.starasia.com/Download/Manual/comemu_starline_pm.pdf) I have tried but couldn't extract the exact bytes to set in the buffer to print, today tested same above code the reciept was empty.

closed 2019-02-22 by JefferyHus