Klemen1337/node-thermal-printer #45: TSP100 (143 LAN) Not printing

blairkelly, 2017-12-16:
Hi there - 
I'm new to thermal printers. The printer I'm trying to use is a Star TSP100IIILAN model (143 LAN). If I specify in the example.js options an interface of "tcp://10.0.1.15:9100", the net interface will successfully connect to the printer. The buffer is sent to the printer. Then there's a data event which returns "#�" and then it disconnects.
 
However, the printer does nothing.

I don't know if this is an issue with node-thermal-printer; I think I'm doing something wrong. 

Perhaps this particular STAR printer model needs different commands sent to it than what are specified in this repo's lib/star.js?

Here's some more info, just in case it helps: I've also been having trouble with the StarIO Linux SDK: I can connect to the printer, get the Parsed Status, and even pop the cash drawer, but I can't print the sample receipt. However, I can print sample receipts without problem using Star's Swift SDK for iOS development. This is consistent across two working printers of the same model. 

Perhaps it's an authentication problem?

Can you point me in the right direction? What do you think I'm doing wrong?

blairkelly, 2017-12-20:
After quite a bit of research, it looks like the Star TSP100/TSP143 LAN is not able to print using Line Mode Commands or ESC/POS from Linux: the solution has been to generate a PDF from HTML and then print the PDF using the node-printer library.

closed 2017-12-20 by blairkelly

msariman-rko, 2019-03-08:
Hi blair,
   I have a TSP100 LAN too, do u have a sample code that u manage to print?

jameslaydigital, 2019-04-01:
Shouldn't have anything to do with Linux.  My coworkers and I discovered the TSP100 works in raster mode, so you can print a raster image then use the raster cut command.  Line-mode commands don't work.

msariman-rko, 2019-04-01:
Is there a sample JAVA code snippets and a jar library for developing raster mode?

Thanks

jameslaydigital, 2019-04-02:
@msariman-rko I don't know. Try printing an image and see if that works - printing an image uses raster commands.  If that's the case, you should be able to bake your whole receipt into an image and send that to the printer...

nipoonp, 2023-01-10:
I made images print work like this using the node-html-to-image library.

```
        const image = await nodeHtmlToImage({
            html: `<html><body><div style="font-size: 24px;">Check out what I just did! #cool</div></body></html>`,
            quality: 20,
        });

        console.log("xxx... image", image);

        await printer.printImageBuffer(image); // Print PNG image

```