Klemen1337/node-thermal-printer #54: printer.openCashDrawer() not opening drawer

nielsrobin, 2018-03-22:
I'm trying to open the cash drawer however nothing is happening. 

Code:
```javascript
var printer = require("node-thermal-printer")

printer.init({
  type: printer.printerTypes.STAR,
  interface: 'printer:Star TSP100 Tear Bar (TSP113)',
  characterSet: 'UTF8'
})    

printer.alignCenter()
printer.isPrinterConnected( function(isConnected){ console.log(isConnected) } )
printer.println("Hello Mr. Santiago")
printer.code128("20101", {
  height: 50,
  text: 1
})
printer.cut()
printer.openCashDrawer()
printer.execute( function(err){ if(err) console.log(err) } )
```
Everything else seems to work though. The printer is a "Star TSP100 Tear Bar (TSP113)" and is connected to the cash drawer via phone-jack (I guess it is called).

Klemen1337, 2018-04-05:
It also depends on the drawer. Some work, some don't....
You can probably get it working by changing the command so it sends a pulse to a different pin
```
CD_KICK_2  : new Buffer([0x1b, 0x70, 0x00]),              // Sends a pulse to pin 2 []
CD_KICK_5  : new Buffer([0x1b, 0x70, 0x01]),              // Sends a pulse to pin 5 []
CD_KICK    : new Buffer([0x1b, 0x07, 0x0b, 0x37, 0x07]),  // Kick the cash drawer
```

closed 2019-01-22 by Klemen1337

whoisnyl, 2022-05-23:
> It also depends on the drawer. Some work, some don't.... You can probably get it working by changing the command so it sends a pulse to a different pin
>
> ```
> CD_KICK_2  : new Buffer([0x1b, 0x70, [...]

Hi @Klemen1337 could you tell where and how to use this command you suggested? Sorry I'm a bit confused. Thank you!