Klemen1337/node-thermal-printer #55: printer.isPrinterConnected() always returns true for USB printer

keerthi158, 2018-03-28:
I have printed the bill using USB printer. But when i turn off the printer, it won't return the "false" value for the printer.isPrinterConnected(). It is always prints the "Print done" on console.
`var printer = require("node-thermal-printer");
    printer.init({
        type: 'epson',
        interface: 'printer:RONGTA RP80 Printer'
    });
    printer.setTextDoubleWidth();
    printer.bold(true);
    printer.println("Hello World!");
    printer.println("********");
 
    printer.isPrinterConnected( function(isConnected){
        console.log(isConnected);
        if(isConnected){
            printer.cut();
                printer.execute(function(err){
                    if (err) {
                        console.error("Print failed", err);
                    } else {
                        console.log("Print done");
                    }
                });
        }else{
            alert("printer is not connected/Off! Please check it!.");
        }
    });`

manueljtejada, 2018-09-05:
@keerthi158 were you able to find a fix for this?

pporcel, 2018-09-21:
Any solution?

closed 2019-02-28 by stale[bot]

mattiabettinsoli98, 2019-08-22:
Any solution?