Klemen1337/node-thermal-printer #56: Start TSP100III wifi printer says connected but cannot print

burakkilic, 2018-04-05:
Hi,

I couldn't make my new Star printer print. here is the code:

    printer.init({
                type: printer.printerTypes.STAR,  // 'star' or 'epson'
                interface: "tcp://192.168.2.8:9100",
                width: 48,                         // Number of characters in one line (default 48)
                characterSet: 'SLOVENIA',          // Character set default SLOVENIA
                removeSpecialCharacters: false,    // Removes special characters - default: false
                replaceSpecialCharacters: true,    // Replaces special characters listed in config files - default: true
                // lineChar: "=",                  // Use custom character for drawing lines
                ip: "localhost",                // Ethernet printing IP
                port: 9001                      // Ethernet printing PORT
            });

            printer.alignCenter();
            printer.println("Hello world");
            printer.cut();
            printer.execute(function (err) {
                if (err) {
                    console.error("Print failed", err);
                    reject(err);
                } else {
                    console.log("Print done");
                    return resolve(true);

                }
            });

It writes "Print Done" to to console but never prints actually. Is there anyone who faced the same problem?

burakkilic, 2018-04-07:
I'm getting 

    Print failed! Reason : Error: connect ECONNREFUSED 192.168.2.8:9100

Also I ping the printer:

    Buraks-MacBook-Pro:fdprintserver burak$ nc -z 192.168.2.8 1-50000
    Connection to 192.168.2.8 port 23 [tcp/telnet] succeeded!
    Connection to 192.168.2.8 port 80 [tcp/http] succeeded!
    Connection to 192.168.2.8 port 515 [tcp/printer] succeeded!
    Connection to 192.168.2.8 port 9100 [tcp/hp-pdl-datastr] succeeded!
    Connection to 192.168.2.8 port 9101 [tcp/bacula-dir] succeeded!
    Connection to 192.168.2.8 port 22222 [tcp/*] succeeded!

Where am I wrong?

keerthi158, 2018-04-13:
Sorry for the delayed response.
The message 'Connection Refused' has two main causes:
- Nothing is listening on the IP:Port you are trying to connect to.
- The port is blocked by a firewall.

Please ping and check whether you get the response from printer. (sending packets and receive it)
Also check any of the device has same IP which is assigned for printer.

lilyyang88, 2018-09-15:
Having the sam problem, how can I tell if the port is blocked or nothing is listening?

philyoosays, 2018-09-18:
I am getting the same on a POSBANK A7 (runs Epson). I can print the first time after I turn the printer off and on but only if the print code is in a separate script as opposed to called by my express server. If I try to run the print code from my express server, I always get isPrinterConnected true but ECONNREFUSED as well. Any attempt to print from my separate script after the first print causes isPrinterConnected to be true but get ECONNREFUSED. Just once though, if it helps locate the problem I got a message just once in my console saying Socket Timed Out

philyoosays, 2018-09-18:
@lilyyang88 @burakkilic  @keerthi158 
So I just solved the ECONNREFUSED issue. The issue seems to stem from the isPrinterConnected method. I commented out that method call and I was able to print freely to the printer. Hope this helps!

lilyyang88, 2018-09-19:
There is no more error messages, but still not printing anything...

closed 2019-02-28 by stale[bot]

Dhoffe, 2021-09-29:
> @lilyyang88 @burakkilic @keerthi158 So I just solved the ECONNREFUSED issue. The issue seems to stem from the isPrinterConnected method. I commented out that method call and I was able to print freely to the printer. Hope this helps!

This worked!