Klemen1337/node-thermal-printer #57: Print Chinese Garbled!

qjdice, 2018-04-19:
I want to print chinese but print out the Garbled.
English Is OK.

this is my code:

`var printers = require("node-thermal-printer");
printers.init({
  type: 'epson',
  interface: 'printer:' + printer.getPrinters()[0].name,
  characterSet:'CHINA'
});
printers.alignCenter();
printers.println(iconv.encode('--------测试打印----------','GBK'));
printers.println(iconv.encode('--------测试打印----------','UTF-8'));
printers.println('--------测试打印----------');
printers.println('hello');

printers.execute( function(err){
	console.dir(err);
});`

alanzhaonys, 2018-04-28:
Hi, your printer needs to support Chinese character.

closed 2018-08-10 by Klemen1337

jevy-wangfei, 2018-10-13:
> I want to print chinese but print out the Garbled.
> English Is OK.
>
> this is my code:
>
> `var printers = require("node-thermal-printer");`
> `printers.init({`
> `type: 'epson',`
> `interface: 'printer:' + [...]

Hi, 

Did you resolve the problem? I have the same problem, and I pretty sure that my printer char code setting is Chinese(BIG5). 

Yours' Jevy Wang

qjdice, 2018-10-16:
> Hi,
>
> Did you resolve the problem? I have the same problem, and I pretty sure that my printer char code setting is Chinese(BIG5).
>
> Yours' Jevy Wang

I modified the source code need a iconv-lite module 
add code to node-thermal-printer/lib/core.js

```
   var fs = require('fs'),
    net = require("net"),
    PNG = require('pngjs').PNG,
    star = require('./star'),
    epson = require('./epson'),
    getInterface = require("../interfaces"),
    iconv = require('iconv-lite');

```
and println:function(){}

```
> println: function(text,isGbk){
    if(isGbk){
      append(iconv.encode(text,'GBK'))
    }else{
      append(text.toString());
    }
    append("\n");
  },
```

used

```
 var printer = require("node-thermal-printer");
 printer.init({
		  type: 'epson',
		  interface: printname,
		});
printer.println('测试打印中文',true);

```