Klemen1337/node-thermal-printer #61: tableCustom - Same character is printed in first and second line, if the text is not accommodate in first line

keerthi158, 2018-06-07:
Hi,
 I have the table with three columns like Column1, Column2, Column3.
If the text of column has more character, it prints the last character of first line in first character of second line ie same character printed twice.
**EXAMPLE:**
**Expected Output :**
       Chicken Fried Rice With Curry    1    100
**Actual Output :**
       Chicken Fried Rice With C    1    100
                                       Curry

In above, the "C" character printer twice. 

If i change the tableCustom function in core.js file as below, it works.

`      if(tooLong){
        secondLineEnabled = true;
        //obj.text = obj.originalText.substring(cellWidth-1);
        obj.text = obj.originalText.substring(cellWidth);
        secondLine.push(obj);
      } else {
        obj.text = "";
        secondLine.push(obj);
      }`

Please comment on this.

jevillC4, 2018-08-24:
@keerthi158  I have left a solution reviews the issues **repeated character**

msedek, 2018-09-13:
```
  // Options: text, align, width, bold
  tableCustom: function(data) {
    var cellWidth = printerConfig.width / data.length;
    var secondLine = [];
    var secondLineEnabled = false;

    for (var i = 0; i < data.length; i++) {
      var tooLong = false;
      var obj = data[i];
      obj.text = obj.text.toString();

      if (obj.width) cellWidth = printerConfig.width * obj.width;
      if (obj.bold) module.exports.bold(true);

      // If text is too wide go to next line
      if (cellWidth < obj.text.length) {
        tooLong = true;
        obj.originalText = obj.text;
        obj.text = obj.text.substring(0, cellWidth-1);
      }

      if (obj.align == "CENTER") {
        var spaces = (cellWidth - obj.text.toString().length) / 2;
        for (var j = 0; j < spaces; j++) {
          append(new Buffer(" "));
        }
        if (obj.text != "") append(obj.text);
        for (var j = 0; j < spaces - 1; j++) {
          append(new Buffer(" "));
        }
      } else if (obj.align == "RIGHT") {
        var spaces = cellWidth - obj.text.toString().length;
        for (var j = 0; j < spaces; j++) {
          append(new Buffer(" "));
        }
        if (obj.text != "") append(obj.text);
      } else {
        if (obj.text != "") append(obj.text);
        var spaces = cellWidth - obj.text.toString().length;
        for (var j = 0; j < spaces; j++) {
          append(new Buffer(" "));
        }
      }

      if (obj.bold) module.exports.bold(false);

      if (tooLong) {
        secondLineEnabled = true;
        obj.text = obj.originalText.substring(cellWidth - 1);
        secondLine.push(obj);
      } else {
        obj.text = "";
        secondLine.push(obj);
      }
    }

    module.exports.newLine();

    // Print the second line
    if (secondLineEnabled) {
      module.exports.tableCustom(secondLine);
    }
  },
```

would be nice if the developer upload the fix to the bug.. this line  `obj.text = obj.text.substring(0, cellWidth-1); `cellWidth does not have the -1 in the original code.. but I have debugged to the point I found that was the culprit.. so requesting the developer to fix the source so everyone keep enjoying the best thermal printer library, cheers

Klemen1337, 2019-02-20:
Duplicate #71

closed 2019-02-20 by Klemen1337