Klemen1337/node-thermal-printer #66: Font size issue

jvaisaonna, 2018-07-30:
Can I have bigger size for the font?
I want the font size is bigger than quad area.

Thank you so much!!!

Klemen1337, 2018-08-06:
@jvaisaonna it is a printer limitation.

closed 2018-08-06 by Klemen1337

jvaisaonna, 2018-08-06:
I found the solution on epson website.
I added following code to epsonConfig.js for customize the font size from size 0 to 7.

> TXT_0           : new Buffer([0x1d, 0x21, 0x00]),
>     TXT_1           : new Buffer([0x1d, 0x21, 0x11]),
>     TXT_2           : new Buffer([0x1d, 0x21, 0x22]),
>     TXT_3           : new Buffer([0x1d, 0x21, 0x33]),
>     TXT_4           : new Buffer([0x1d, 0x21, 0x44]),
>     TXT_5           : new Buffer([0x1d, 0x21, 0x55]),
>     TXT_6           : new Buffer([0x1d, 0x21, 0x66]),
>     TXT_7           : new Buffer([0x1d, 0x21, 0x77]),