Klemen1337/node-thermal-printer #70: Printing euro sign (€)

Miguel-Frazao, 2018-08-20:
I have been trying to add the euro sign in the output of the printer but with no success, is it possible to integrate it? If yes, How? https://www.fileformat.info/info/unicode/char/20ac/index.htm

Klemen1337, 2018-08-21:
What printer do you have?

Miguel-Frazao, 2018-08-21:
I have a Birch BP T3, and it's configured like an epson:

```
...
printer.init({
            type: 'epson',
            interface: 'tcp://...'
...
```

It's was printing with the old software that we had, made in .NET (c#), but we decided to upgrade it to this one.

kozmoz, 2018-09-26:
Tested with an **Epson TM-T88V**

Before you're able to print the Euro currency sign €, set the code page to a page that supports the Euro. e.g. code page ISO8859-15 (code 40):

`printer.add(new Buffer([0x1b, 0x74, 40]));`

After that you can print the Euro currency sign by sending code 164 directly to the printer:

`printer.add(new Buffer([164]))`

Or let the library handle the Euro currency sign automatically. Here is a complete example, that configures the conversion of the UTF-8 Euro currency sign to character 164 on code page 40:

```javascript
var printer = require("../node-thermal-printer");
printer.init({
    type: printer.printerTypes.EPSON,
    interface: 'tcp://10.0.1.23:9100',

    // Replace the Euro sign in strings by code 164.
    replaceSpecialCharacters: true,
    extraSpecialCharacters: {'€': 164}
});

// Set the code page to ISO8859-15 (Western Europe with Euro),
// so we can print the Euro.
printer.add(new Buffer([0x1b, 0x74, 40]));

printer.println('Print Euro: €€€€€€');
printer.newLine();
printer.newLine();

printer.cut();
printer.execute();
```

mo-g, 2018-10-10:
Worth a patch to allow codepage to be set more cleanly? Currently, one needs to use manual ESC/POS commands over Buffer, as you describe it (thanks for doing so, by the way!).

It would make sense to me to have the codepage be settable by name, like setting the characterset - and possibly setting some special character values automagically based on the codepage chosen.

Also, small criticism of your code - better to use codepage 19 and character 213 - codepage 40 breaks `drawLine().`

kozmoz, 2018-10-11:
Was indeed thinking about creating a PR for setting the codepage. Or even better, let the library do the translation from UTF-8 to the right character completely automatic. It's 2018 though! Why should the software that uses this library still use characters sets and codepages to print different characters. imho it doesn't make sense.

Thanks for mentioning codepage 19. that codepage doesn't break the `drawLine()` function.

kozmoz, 2018-10-12:
`escpos-php` does something similar. 

At initialization, the the code page needs to be selected. After that, it tries to match all UTF characters to this code page. Characters without a match are printed as space. So eg if the user selects a code page that supports the Euro currency symbol, it will make the translation from Euro in text to the correct code automatically. No need to find and print the character code for Euro.

https://github.com/mike42/escpos-php/blob/development/src/Mike42/Escpos/resources/capabilities.json

mo-g, 2018-10-12:
What would be the latency cost of picking the code page on a character by character basis?

texttexttextspecialcharactertext-changecodepage-specialcharacter-changecodepage-specialcharacter-changecodepage-specialcharactertexttexttext. There could be a complete codepage reverse lookup table - on a unicode character outside ASCII, the library checks if it's in the current codepage and if not; changes to one that has it.
```
{
    "€": {
        "19": "213",
        "40": "164"
    }
}
```
That way, users never need to deal with any of the character set or code page junk - they just send unicode strings and the library deals with it.

kozmoz, 2018-10-14:
I've created a Proof of Concept: https://github.com/kozmoz/node-thermal-printer/tree/find-codepage-automatically-iconv

Works like a charm for my Epson.

Cattpot, 2018-10-15:
> Tested with an **Epson TM-T88V**
> 
> Before you're able to print the Euro currency sign €, set the code page to a page that supports the Euro. e.g. code page ISO8859-15 (code 40):
> 
> `printer.add(new Buffer([0x1b, 0x74, 40]));`
> 
> After that you can print the Euro currency sign by sending code 164 directly to the printer:
> 
> `printer.add(new Buffer([164]))`
> 
> Or let the library handle the Euro currency sign automatically. Here is a complete example, that configures the conversion of the UTF-8 Euro currency sign to character 164 on code page 40:
> 
> ```js
> var printer = require("../node-thermal-printer");
> printer.init({
>     type: printer.printerTypes.EPSON,
>     interface: 'tcp://10.0.1.23:9100',
> 
>     // Replace the Euro sign in strings by code 164.
>     replaceSpecialCharacters: true,
>     extraSpecialCharacters: {'€': 164}
> });
> 
> // Set the code page to ISO8859-15 (Western Europe with Euro),
> // so we are able to print the Euro currency sign.
> printer.add(new Buffer([0x1b, 0x74, 40]));
> 
> // Print the Euro currency sign from a string.
> printer.println('Print Euro: €€€€€€');
> printer.newLine();
> 
> // Print the Euro currency sign by sending code 164 to the printer.
> printer.add(new Buffer([164]));
> printer.newLine();
> printer.newLine();
> 
> printer.cut();
> printer.execute();
> ```

which characterSet are you using in the init?

Cattpot, 2018-10-15:
> > which characterSet are you using in the init?
> > still not working :(
> 
> Just using the default charset set by the library, but afaik it doesn't really matter, because later on we set the code page.
> 
> Actually, only the code below is necessary to print an Euro sign. What do you see? What's the brand of the printer?

TM-20ii
Im using your example with my ip, I get a ñ instead of €.
How to set a "code page"?

kozmoz, 2018-10-15:
Somehow the code page of the printer doesn't change, it's stuck in code page 0, hence the ñ. 

The code below sets code page to 40 (ISO8859-15 encoding):
```javascript
printer.add(new Buffer([0x1b, 0x74, 40]));
```

Code page 19 for example also contains an Euro sign:
```javascript
printer.add(new Buffer([0x1b, 0x74, 19]));
printer.add(new Buffer([213]));
printer.execute();
```

Cattpot, 2018-10-15:
Thanks very much!

``` js
// Set the code page to ISO8859-15 (Western Europe with Euro),
// so we are able to print the Euro currency sign.
printer.add(new Buffer([0x1b, 0x74, 40]));
```

Somehow the line was at the end.
Put it to the front. Now the € is working.

`printer.drawLine()`
is now writing ÄÄÄ.

mo-g, 2018-10-15:
> Also, small criticism of your code - better to use codepage 19 and character 213 - codepage 40 breaks `drawLine()`.

closed 2019-01-22 by Klemen1337