Klemen1337/node-thermal-printer #75: Chinese Compatiblity

pholmesp, 2018-09-19:
Hi, I have been trying to print receipts in Chinese with this module, and I have tried to set the parameter "characterSet" to "CHINESE" or "MANDARIN" but none of them worked. Could anybody help? Thanks.

sineth855, 2018-11-06:
I found the sample problem as u, how is your research?

closed 2019-02-28 by stale[bot]

wesleyxiao, 2019-12-20:
+1, i try to print chinese on tm t88v thermal print also show Garbled, any update?

pholmesp, 2019-12-20:
> +1, i try to print chinese on tm t88v thermal print also show Garbled, any update?

I got the problem solved by doing a little modification on the package source code. I will post the solution later today.

wesleyxiao, 2020-04-05:
> > +1, i try to print chinese on tm t88v thermal print also show Garbled, any update?
> 
> I got the problem solved by doing a little modification on the package source code. I will post the solution later today.

:) great any update here?

pholmesp, 2020-04-05:
> > > +1, i try to print chinese on tm t88v thermal print also show Garbled, any update?
> > 
> > 
> > I got the problem solved by doing a little modification on the package source code. I will post the solution later today.
> 
> :) great any update here?

Sorry for the epic delay.

Please go to node-Modules/node-thermal-printer/lib/core.js and replace function print with code below.

`print(text,isGbk) {
    text = text || "";
    if(isGbk){
      this.append(iconv.encode(text,'GBK'))
    }else{
      this.append(text.toString());
    }
    // append("\n");
    // text = text || "";
    // this.append(text.toString());
  }`

Also add iconv-lite if needed.

wesleyxiao, 2020-04-05:
Thank you for information, it still not working.
at core.js i add following code
`  
println(text,isGbk) {
    text = text || "";
    if(isGbk){
      this.append(iconv.encode(text,'GBK'))
    }else{
      this.append(text.toString());
    }
    // append("\n");
    // text = text || "";
    // this.append(text.toString());
  }`

i am also add iconv-lite, and also print as printer.println("打印错误，没有菜品", true);

still show Garbled, do you use same printer tm t88v?

jevy-wangfei, 2020-04-23:
> Thank you for information, it still not working.
> at core.js i add following code
> ` println(text,isGbk) { text = text || ""; if(isGbk){ this.append(iconv.encode(text,'GBK')) }else{ [...]
I resolved this issue by referencing this issue: https://github.com/song940/node-escpos/issues/88

```
// Select Kanji character mode
await printer.print('\x1c\x26') // FS &
// Cancel Kanji character mode
await printer.print('\x1c\x2e') // FS .
```