Klemen1337/node-thermal-printer #78: how can I detect paper empty

ludi81, 2018-10-06:
hi,

I used this library to print an invoice (within an exhibition) with an epson printer. everything works fine. Sometimes the paper gets empty and we need some time to replace it (e.g. 1 hour). 

After replacing the paper it will print  plenty of invoices (more then generated jobs). It seams, that the library starts one job multiple times.
Is it possible to clear all printer jobs? Or can I ask the printer if the paper is empty?

kind regards

SamuelMorales, 2018-12-06:
Im having the same issue, looking a fork of this proyect I saw that they supported the getStatus and ABS method that query for paper issues and status changes, they where right after the isPrinterConnected() method, but it seems that epson deprecated the instruction on wich this method was based on. I think this is the reason why this library does not support it anymore.  Anyways, looking at epson documentation, it seems to be other instructions to query the printer for paper status, could you implement those?. I've tried but no luck yet.

ludi81, 2018-12-06:
Hi,
I changed the file.js file. They use write-file-queue which retries  the printing job several times.

`var writeFile = require("write-file-queue")({
  retries : 1000, 						    // number of write attempts before failing
  waitTime : 200 					        // number of milliseconds to wait between write attempts
  //, debug : console.error 			// optionally pass a function to do dump debug information to
});`

I used 
`var fs = require("fs");`
instead.

If the write method returns an error, I know that another job could not be finished (paper empty)

In my case only one print job will be printed after replacing the paper.

ludi81, 2018-12-07:
The idea is good, but in my opinion it will not work.
Your idea will only work if the paper gets empty in the first dummy print. If it gets empty during the real print you get stuck. I created a timeout (e.g. 2 second). If the job is not done during this time I show a message on the screen (to change the paper).
My solution is not perfekt, but it does the job. If you finde a better way, please let me know.

SamuelMorales, 2018-12-07:
Yes, your solution is better because you signal the error as soon as the printer runs out of paper. I think a combination of yours and mine will be most complete in the sense that mine will prevent any print being done if the printer is out of paper at any time e.g. just after lunching the program with no paper in the printer and no jobs in queue, and yours will make the signal just after the paper runs out during a print.

ludi81, 2018-12-07:
Give it a try. I am not sure if yours will prevent any prints if the printer is out of paper.

for example:
1) you print the dummy line --> paper is not empty
2) you print the real printing --> paper gets during this print empty and the code will not return.
3) a person replaces the paper
4) your real printing will be printed

In this case your dummy line has not the effect as you wanted and described. It only works if the paper gets empty during the dummy line. The probability that this happens was in my scenario quite small (my real print hast about 20cm, just a line about 0,5cm, that's 40:1).

I tried several different ways and found no perfect solution. Maybe it works with a different interface (e.g. the network interface).

SamuelMorales, 2018-12-07:
I might have missunderstood what you ment with the timeout you said before. The current print job wont be done if a previous job is in queue and it is preventing the current job from starting, so thats when the write method you said before returns error? So if out of paper in the middle of a printing, there's no error?. In a nutshell, can we check for running out of paper as soon as it happens or we depend of a print job that wont start because of no paper with a previous job in queue?

ludi81, 2018-12-07:
Yes, I can detect no paper after a couple of seconds (It is not necessary to start a next job). I will try to explain it in other words:

A) you have to change the file.js code from var writeFile = require("write-file-queue") to use var fs = require("fs"); Otherwise the library tries to print the printing several times and after filling up the paper you will print a couple of prints, instead only one.

B) My printing is done like this.
1) before starting the  printer.execute(); job I startet a timer that invokes a function after a couple of seconds (e.g. 2 second).
2) start printer.execute();
3) if the code returns from printer.execute() I cancel the timer from point 1

If the timer finished and invokes the function it means, that the printer is blocked (e.g. paper is empty or printer is open). In this case I show a message to the user, that he/she has to check the printer and replace the paper if necessary.

SamuelMorales, 2018-12-07:
Thank's for the explanation!, now I understand what you ment before. I will give it a try and see how it goes. If I find other way I'll let you know.

closed 2019-01-22 by Klemen1337