Klemen1337/node-thermal-printer #80: Cannot print Image

KakarN, 2018-10-14:
I am trying to print an image using this library and the node-print library.

```
var thermal_printer = require("node-thermal-printer");
var printer = require("printer");
var epson = printer.getPrinter('EPSON_TM_T82_S_A');

thermal_printer.init({
	type: 'epson'
})

thermal_printer.println("Hello world");
thermal_printer.printImage('./assets/main-logo.png', function (done) {
	thermal_printer.execute(function (err) {
		if (err) {
			console.error("Print failed", err);
		} else {
			console.log("Print done");
		}
	});
});

printer.printDirect({
	data: thermal_printer.getBuffer(),
	printer: epson.name,
	type: "RAW",
	success: function (job_id) {
		console.log('OK :' + job_id);
	},
	error: function (err) {
		console.error(err);
	}
});
```

But it does not print an image, instead I am getting an error:

```
kakar@Apples-MacBook-Pro electron-pos $ node image.js
OK :80
fs.js:653
  binding.open(pathModule.toNamespacedPath(path),
          ^

TypeError: path must be a string or Buffer
    at Object.fs.open (fs.js:653:11)
    at Object.fs.writeFile (fs.js:1300:6)
    at self.running (/Users/kakar/electron-apps/electron-pos/node_modules/write-file-queue/index.js:53:12)
    at Immediate.<anonymous> (/Users/kakar/electron-apps/electron-pos/node_modules/dank-do-while/index.js:27:7)
    at runCallback (timers.js:756:18)
    at tryOnImmediate (timers.js:717:5)
    at processImmediate [as _immediateCallback] (timers.js:697:5)
```

shyammtp, 2018-11-22:
Try to Execute the print inside the image Callback.
```
printer.printImage('./logo.png', function(done){ 
   if(done){
       printer.execute(...)
   }
});
```

SloInfinity, 2019-02-20:
As you can see from the error report, the issue is clearly with opening the image file, and the stack trace you provided does not include any mentions on the node-thermal-printer lib. 
If you have more data to provide, please reopen the ticket and provide the data needed.

closed 2019-02-20 by SloInfinity