Klemen1337/node-thermal-printer #86: Not Working with epson printer

shajunr88, 2018-11-30:
I am using EPSON TM-T82 thermal printer.When connected using USB,It shows connected message in my node app.But when trying to print,it does not print anything.I need to know whether this library is compatible only with intelligent printers (ie. TM-T82 i versions).

anujmv, 2018-12-04:
It works fine. I use node printer to detect printer and use this library to generate raw file.

danielgoodwin, 2019-01-14:
I had similar issue with EPSON TM7011 on USB on OSX, so following may be useful for someone else
Steps I did to resolve
- Installed EPSON driver  see https://download.epson-biz.com/modules/pos/index.php?page=single_soft&cid=4226&scat=33&pcat=3
- from command line to get CUPS web interface operational ( cupsctl WebInterface=yes )
- connect to http://localhost:631
- setup printer from Administration
- installed 'node-printer' -  npm install printer
- set interface of examples/example.js to 'printer:EPSON_TM0711'  (see CUPS printer list on web interface for name to use)
- ran example (node examples/example.js ) and output worked.
- Make sure to disable cupctl web interface when done (cupsctl WebInterface=no)

closed 2019-05-01 by stale[bot]