Klemen1337/node-thermal-printer #88: Node Thermal with VueJs Integration

igorbag, 2019-01-14:
Hello everyone.

Is it possible to integrate this lib with the VueJs?

Has anyone used anything related?

kozmoz, 2019-01-14:
It needs to run on Node, so it can access the printer. It is not running in the browser, like Vue.js is.

I use AngularJS as the front-end layer within an Electron app (the front-end part runs in the Chrome layer). node-thermal-printer is running in the Node proces of the app and the frontend communicates with it using events.

https://electronjs.org

igorbag, 2019-01-15:
Hello, @kozmoz!!

thanks for help.

In my case, i use ElectronJs and VueJs to Front-end layer. 

Could you tell me how to integrate the printer with the Electron and later how to use it in VueJs or Angular?

kozmoz, 2019-01-15:
From the frontend you can use the ipcService to send messages to the Node process.
The pinter library lives in the Node layer. In short:

Front-end JS (Angular or Vue):
```javascript
electron = require('electron');
electron.ipcRenderer.send('printOrder', ...args);
```

And in the main.js file (Node):
```javascript
const {ipcMain} = require('electron');
ipcMain.on('printOrder', (event, order) => {
    printServices.printOrder(order)
        .then(() => event.sender.send(`printOrderSuccessResponse`, true))
        .catch(error => event.sender.send(`printOrderErrorResponse`, error));
});
```

The printService looks like this:
```javascript
const printer = require('node-thermal-printer');
const printerConfig = {
    type: printer.printerTypes.EPSON,
    interface: `tcp://${config.printerIp}:` + (config.printerPort || 9100),
};
printer.init(printerConfig);
printer.println(' Printer successfully configured ');
printer.execute();
```

I left out the code for the Promises and also didn't include the listeners for the "printOrderSuccessResponse" to detect if printing was successful, but I think you will get the idea.

Good luck!

igorbag, 2019-01-15:
Awesome @kozmoz , thanks a lot for your help.

closed 2019-01-22 by Klemen1337