Klemen1337/node-thermal-printer #93: electron build  32 bit fail to run

MarioRozic, 2019-02-18:
So I am using this module for the [Electron](https://github.com/electron/electron) to print some stuff with POS printer, and it works fine when I run it in dev mode, but as soon as I build a 32bit version of that  code for Windows, I got an error pointing that this module is missing, which is strange because 64Bit version works perfectly.

[image]

Here is command that I use for building app:
`"package-win32": "electron-packager . --overwrite --platform=win32 --arch=ia32 --prune=true"`

Klemen1337, 2019-02-19:
This package does not need to be rebuilt. Looks like a different npm package.

MarioRozic, 2019-02-19:
Actually, the app is passing the build, this happens only when I execute `exe` file made by electron-packager!

SloInfinity, 2019-02-20:
@MarioRozic you opened the ticket on the wrong package lib. Based on the error message you can clearly see that the issue is not the Node-thermal-printer library but rather the Node-printer. You might want to check this issue https://github.com/tojocky/node-printer/issues/167 which is a similar issue you are having.

closed 2019-02-20 by SloInfinity