Klemen1337/node-thermal-printer #94: Print multiple printers

zeoliveirajunior, 2019-02-20:
Hi,

I have 3 printers on my network, and i need to print on 3 at same time. I check the code of thermal print, and i see that the lib shares the same buffer, and the only solution to print in more than one printer is to create a semaphore, but this solution creates a slowness in the print process.

Is there anyway to print in more then one printer at same time?

Thanks in advance

Klemen1337, 2019-02-21:
Try it like this.

```
var printer = require("../node-thermal-printer");
var printer2 = require("../node-thermal-printer");
var printer3 = require("../node-thermal-printer");

// Setup printers
printer.init({
  type: printer.printerTypes.EPSON,
  interface: 'tcp://192.168.1.1'
});

printer2.init({
  type: printer.printerTypes.EPSON,
  interface: 'tcp://192.168.1.2'
});

printer3.init({
  type: printer.printerTypes.EPSON,
  interface: 'tcp://192.168.1.3'
});

// Add some text
printer.println("This is a test");

// Execute
let buff = printer.getBuffer();
printer.execute();
printer2.raw(buff);
printer3.raw(buff);
```

Klemen1337, 2019-02-27:
See [multiple printers example](https://github.com/Klemen1337/node-thermal-printer/blob/master/examples/multiple-pritners.js)

closed 2019-02-27 by Klemen1337