Klemen1337/node-thermal-printer #97: Error: The specified module could not be found.

shamilbuzz, 2019-03-13:
**When I try to print it says the following, but the file is in specified path already.
OS: Windows 10, node  v10.15.1, npm v6.8.0**

`node_modules\electron-printer\build\Release\electron-v0.36-win32-x64\node_printer.node
    at process.module.(:9080/anonymous function) [as dlopen] (ELECTRON_ASAR.js:160:31)
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:722)
    at Object.module.(:9080/anonymous function) [as .node] (ELECTRON_ASAR.js:160:31)
    at Module.load (internal/modules/cjs/loader.js:602)
    at tryModuleLoad (internal/modules/cjs/loader.js:541)
    at Function.Module._load (internal/modules/cjs/loader.js:533)
    at Module.require (internal/modules/cjs/loader.js:640)
    at require (internal/modules/cjs/helpers.js:20)
    at Object.<anonymous> (VM1396 D:\works\office\nkg-kenya\nkg-kenya-desktop\node_modules\electron-printer\printer.js:4)
    at Object.<anonymous> (VM1396 D:\works\office\nkg-kenya\nkg-kenya-desktop\node_modules\electron-printer\printer.js:6)
    at Module._compile (internal/modules/cjs/loader.js:693)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:704)
    at Module.load (internal/modules/cjs/loader.js:602)
    at tryModuleLoad (internal/modules/cjs/loader.js:541)
    at Function.Module._load (internal/modules/cjs/loader.js:533)
    at Module.require (internal/modules/cjs/loader.js:640)
    at require (internal/modules/cjs/helpers.js:20)
    at Object.<anonymous> (VM1395 D:\works\office\nkg-kenya\nkg-kenya-desktop\node_modules\electron-printer\lib\printer.js:6)
    at Object.<anonymous> (VM1395 D:\works\office\nkg-kenya\nkg-kenya-desktop\node_modules\electron-printer\lib\printer.js:319)
    at Module._compile (internal/modules/cjs/loader.js:693)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:704)
    at Module.load (internal/modules/cjs/loader.js:602)`

Have anyone faced this issue? Need help 😢

Klemen1337, 2019-03-18:
Please open issue electron-printer npm module

closed 2019-03-18 by Klemen1337

madrojudi, 2019-06-25:
Hello, Please have you got the solution? I have same problem one month ago.
Thanks