Klemen1337/node-thermal-printer #98: fs module not found

yalim, 2019-04-01:
Hi, I am trying to use the plugin for EPSON printer for an Android app but I receive follwing error:

```
System.err: Error: Building UI from XML. @file:///app/app-root.xml:4:9
System.err:  > com.tns.NativeScriptException: Failed to find module: "fs", relative to: app/tns_modules/
```
my node version is v10.15.3

I thought about removing fs requirement since I don't intend to print images but later I noticed it is also required for isPrinterConnected(). Is there a way to solve this issue? 

Thanks

yalim, 2019-04-04:
I ended up removing all require('fs') and related functions but it still gives the same error. Do you know if any of the dependencies also need 'fs' to run?

Klemen1337, 2019-05-10:
fs is part od nodejs

closed 2019-05-10 by Klemen1337

robbtt, 2019-11-08:
@Klemen1337 so what is the fix for this? Thanks!