Klemen1337/node-thermal-printer #99: Star TSP600 series: printer fires several minutes after issuing printImage()

jameslaydigital, 2019-04-01:
On Windows 10, using the TCP interface, I'm trying to issue the following:

    let printer = new ThermalPrinter({
        type: PrinterTypes.STAR,
        interface: 'tcp://10.1.10.40'
    });
    await printer.printImage('z.png');
    console.log(printer.getBuffer().length); // 104623 <- comes back almost immediately
    printer.cut();
    printer.execute();

What happens is it prints about one dot-row and then several minutes later while I'm getting coffee or something it spits out the rest of the image.  Any idea what could be happening?

jameslaydigital, 2019-04-02:
I think I've made some progress.  If I try to print a smaller image, it will print entirely and even perform the following cut command.  Once the image reaches a certain size it will only partially print the image until I either wait or push more raster commands through.  It will not process any line-commands afterwards until I restart the printer.

jameslaydigital, 2019-04-11:
I've discovered a workaround.  If you send the buffer down the line in chunks instead of all at the same time, it works.  For instance, instead of printer.execute(), I'll manually push 2048-byte chunks from printer.getBuffer() about 5ms apart down a manual tcp connection to the printer.  Not sure if this should be closed, unless I'm the only one who has had this issue.

Klemen1337, 2019-06-27:
@jameslaydigital good work around. Network printing might have to be done like that.

fnaquira, 2019-10-18:
The problem is that you are not giving the callback function.
```
printImage: function(image, callback){
```
As you can see, **printImage** needs a callback, I was having that issue and solved adding that.
Good luck!

CHOMNANP, 2019-12-03:
> ```
> printImage: function(image, callback){
> ```

Hi @fnaquira, can you share the code that work for your case?
I cannot figure out how this would work.

fnaquira, 2019-12-04:
It was hard for me to understand that this function had to read a file, then print it and voilá!
The problem is the original code, it is not a promise, it's just a callback function and you are putting **await** in front of it.
I had to create this:
```
const printImage = function(imgPath,printer){
	return new Promise((resolve,reject) => {
		printer.printImage(imgPath, function(resp) {
			console.log('img printed', resp);
			resolve();
		});
	});
}
printImage('assets/logo.png',printer);
```
Now, when I call the printImage function, it really waits and everything goes right

Klemen1337, 2019-12-05:
@fnaquira you might be using an older version of the node-thermal-printer.
printImage function is async and has no callback.

closed 2020-05-07 by stale[bot]