Klemen1337/node-thermal-printer #103: win10 Cannot print large image，it will be garbled

Anberm, 2019-04-27:
I used win10 system to run examples/image.js on the 80mm thermal printer, and replaced olaii-logo-black-small-png with olaii-logo-black-png, which would print garbled code

Klemen1337, 2019-06-27:
@Anberm what is the name of the thermal printer, what type of printer is it (eprson or star)?

Anberm, 2019-06-27:
@Klemen1337 EPSON ,I printed it by slicing up the picture into small pictures, and it worked

Klemen1337, 2019-06-28:
@Anberm Can you tell me how you did it?

Anberm, 2019-07-01:
@Klemen1337 This is a problem of node-printer, so I will provide a reference solution：
 var dataBuffer = new Buffer(base64Data, 'base64');

  let PNG = require('pngjs').PNG;
  var png = PNG.sync.read(dataBuffer);
  var amp = 200;
  var dpi = parseInt(png.height / png.width);
  var pwidth = png.width + amp;
  var pheight = png.height + amp * dpi;
  var wh = 10;
  var len = pheight % wh ? Math.floor(pheight / wh) + 1 : pheight / wh;
  var remainder = pheight % wh;
  for (let x = 0; x < len; x++) {
    var height = x + 1 === len && remainder ? remainder : wh;
    var data = await sharp(dataBuffer)
      .resize(pwidth, pheight)
      .extract({
        left: 0,
        top: x * wh,
        width: pwidth,
        height: height,
      })
      .toBuffer();
    await printer.printImageBuffer(data);
  }
  printer.execute();

closed 2019-10-16 by stale[bot]

peteruithoven, 2023-10-11:
For future reference, with syntax highlighting:
``` js 
var dataBuffer = new Buffer(base64Data, "base64");

let PNG = require("pngjs").PNG;
var png = PNG.sync.read(dataBuffer);
var amp = 200;
var dpi = parseInt(png.height / png.width);
var pwidth = png.width + amp;
var pheight = png.height + amp * dpi;
var wh = 10;
var len = pheight % wh ? Math.floor(pheight / wh) + 1 : pheight / wh;
var remainder = pheight % wh;
for (let x = 0; x < len; x++) {
  var height = x + 1 === len && remainder ? remainder : wh;
  var data = await sharp(dataBuffer)
    .resize(pwidth, pheight)
    .extract({
      left: 0,
      top: x * wh,
      width: pwidth,
      height: height,
    })
    .toBuffer();
  await printer.printImageBuffer(data);
}
printer.execute();
```