Klemen1337/node-thermal-printer #109: Updating to 4.1 error (webpack_require__(...) is not a function)

infinitetrinity, 2019-05-20:
Updating to 4.1 gives me this error

````
Uncaught TypeError: __webpack_require__(...) is not a function
    at eval (index.js?acf9:144)
    at Object../node_modules/iconv-lite/lib/index.js (app.js:5774)
    at __webpack_require__ (app.js:724)
    at fn (app.js:101)
    at eval (core.js?1bf4:3)
    at Object../node_modules/node-thermal-printer/lib/core.js (app.js:7446)
    at __webpack_require__ (app.js:724)
    at fn (app.js:101)
    at eval (node-thermal-printer.js?a3e2:1)
    at Object../node_modules/node-thermal-printer/node-thermal-printer.js (app.js:7567)
````

Klemen1337, 2019-06-27:
@infinitetrinity I need more informations about this error. This might help you [https://stackoverflow.com/questions/40873599/typeerror-webpack-require-i-is-not-a-function](https://stackoverflow.com/questions/40873599/typeerror-webpack-require-i-is-not-a-function)

infinitetrinity, 2019-07-22:
Update on this. This is a issue with iconv-lite when use for electron app.
Check here https://github.com/ashtuchkin/iconv-lite/issues/205 for more info.

infinitetrinity, 2019-07-22:
Closed by accident. Can @Klemen1337 check on the link i mentioned and see if there's any workaround for us who use this for electron vue app? Thank you

kenguish, 2019-11-26:
The problem has to do with iconv-lite, so the only way to "fix" would be to remove the iconv-lite dependency.

Here is my get around, using Vue CLI 3 as suggested https://github.com/ashtuchkin/iconv-lite/issues/204#issuecomment-459221992
At your node_modules/iconv-lite/lib/index.js, at line 144 and 148, add ".js" to the require like this:
<pre>if (nodeVerArr[0] > 0 || nodeVerArr[1] >= 10) {
        require("./streams.js")(iconv);
}

// Load Node primitive extensions.
require("./extend-node.js")(iconv);</pre>

Klemen1337, 2019-12-25:
@kenguish @infinitetrinity I will update the iconv-lite version to v0.5.0 (includes the fix)

closed 2020-01-07 by Klemen1337

infinitetrinity, 2020-05-03:
> @kenguish @infinitetrinity I will update the iconv-lite version to v0.5.0 (includes the fix)

Hi. I'm still encountering this issue with the latest release.

commit 179ab540 (2020-01-07): - updated iconv-lite from v0.4.24 to v0.5.0 closes #109,
- added example to readme,
- added no driver set error