Klemen1337/node-thermal-printer #110: VIETNAMESE character set doesn't work

manhnguyenquoc, 2019-06-03:
Hi @Klemen1337 , thank you for building this module. I'm from Vietnam and I'm trying to print Vietnamese words but I got the font error paper after printing.

For example: 
I would like to print "Công ty Cổ phần Viễn thông VNNET" and here is what I get:
[image]

My code:
[image]

Could you please tell me how to fix it. Thank you very much.

closed 2019-09-18 by stale[bot]

MY020537, 2019-10-08:
@manhnguyenquoc Were you able to fix this ?

manhtnt, 2023-03-06:
I got the same problem. Can anyone help us?