Klemen1337/node-thermal-printer #113: Type error on typescript

asnawisaharuddin, 2019-06-14:
Unable to compile using typescript.

Error :
`node_modules/node-thermal-printer/node-thermal-printer.d.ts:18:14 - error TS2314: Generic type 'Promise<T>' requires 1 type argument(s).`

Probably need to specify return type of Promise<void | any>.

hadeeb, 2019-06-21:
Type-definitions are out of sync with the changes in `v4`.

@Klemen1337 do you mind if I take up updating type-definitions?

chunkitmax, 2019-06-25:
I have updated for typescript but I also renamed class printer and enum printerTypes.
Plase check [https://github.com/chunkitmax/node-thermal-printer](https://github.com/chunkitmax/node-thermal-printer)

hadeeb, 2019-06-27:
@chunkitmax , I noticed you have multiple changes in your fork.
So, I created a [PR](https://github.com/Klemen1337/node-thermal-printer/pull/115) with type definition updates.
Will please you go through it?

chunkitmax, 2019-06-27:
@hadeeb, I did not check those missing methods so your updates are more complete. I think your PR is good to be merged.

Klemen1337, 2019-06-27:
I don't know how i missed this, but I worked on it today 😄

PR #115 (2019-06-28): Update type definitions
closes #113
closed 2019-06-28 by Klemen1337