Klemen1337/node-thermal-printer #117: Epson TM-T88III via /dev/ttyUSB0

skerit, 2019-07-17:
Last week I got a new Epson TM-T20II to work with node-thermal-printer over an ethernet connection. It really does work great.

But now I have an older device, an Epson TM-T88III with a serial-port connection. I'm ussing a serial-port-to-usb device to connect it to my computer, so I'm trying to connect to `/dev/ttyUSB0` like this:

```javascript
let printer = new ThermalPrinter({
	type: PrinterTypes.EPSON,
	interface: '/dev/ttyUSB0',
});

// This is always true, even when the printer is off. I think it only tests if the /dev/ttyUSB0 file exists?
let isConnected = await printer.isPrinterConnected();

printer.alignCenter();
printer.println("Hello world!");

printer.cut();

try {
	let execute = printer.execute()
	let result = await execute;
	console.error("Print done!", result);
} catch (error) {
	console.log("Print failed:", error);
}
```

The printer does absolutely nothing.
This script eventually times out with this error:

```
Print failed: { [Error: ESPIPE: invalid seek, write] errno: -29, code: 'ESPIPE', syscall: 'write' }
```

I added my user to the required groups, set the correct permissions on the `/dev/ttyUSB0` file and all, but still nothing happens.

I saw `node-thermal-printer` was tested on a TM-T88V, so I was kinda hoping it would just work with an earlier version of that. Is it too old? Or is something going wrong with the connection to the serial port somehow?

tashreek, 2019-07-18:
I am experiencing the same issue except I need to integrate with both USB and serial printers. The library indicates that you should use the interface parameter to pass through the name of the printer e.g. printer:Epson-TM-BA-Thermal which works for me. This however requires the https://www.npmjs.com/package/printer module but it has a ton of build errors and hasn't been updated in 3 years. This isn't a feasible solution as my project can no longer build.

skerit, 2019-07-18:
Ok, I finally got it to work!

In order to connect the printer to a computer via the serial port you need a nullmodem cable. That was my first issue. I bought a little adapter (Null Modem Sub-D 9 pin male » female) which turns an existing, regular serial cable into a nullmodem cable. From there on out, I was able to send commands to the printer!

The other problem is in the "file" implementation of node-thermal-printe. It uses the `write-file-queue` dependency to write to the file, which in turn uses `fs.writeFile()` to actually write the buffer to the file. But `/dev/ttyUSB0` is not a normal file and is not seekable. 

For it to work I had to change the `execute` method of the `file.js` module to use `fs.appendFile()`:

```javascript
  async execute(buffer) {
      return new Promise((resolve, reject) => {

        fs.appendFile(this.path, buffer, function(e) {

          if (err) {
            reject(err);
          } else {
            resolve('Print done');
          }
        });
    }
```

I didn't bother implementing a new `write-file-queue`-like implementation.

I also tried the `serialport` module, which is an even more appropriate way to handle serial connections, but maybe that's a bit overkill in this case?

skerit, 2019-07-18:
@tashreek Using the `printer:` route might seem like a good idea, especially on windows, but on Linux it seems more like a pain. I couldn't get neither of my printers to work properly via CUPS. A direct connection is way faster and less error prone as it just cuts out the CUPS middle-man.

skerit, 2019-07-18:
Well, unfortunately I only have an RS232 and an ethernet printer to test on, but if issue #58 is any indication it should all work the same way: just use it as a file, but with the `appendFile` modification.

The easiest way to test if this would work is to simply echo to the device file, like:

```bash
echo "Testing!" >> /dev/ttyUSB0
```

Other device files could be `/dev/usb/lp0` for usb printers? Don't know where parallel ports would go...

closed 2019-11-02 by stale[bot]