Klemen1337/node-thermal-printer #120: Cant Find Module : node-thermal-printer

MaheshEx1990, 2019-09-23:
Installed plugin as shown but it not compiling
Throwing error Can't Find Module: node-thermal-printer

Klemen1337, 2019-09-25:
Please provide more information.

closed 2019-11-19 by Klemen1337

SimantoR, 2019-12-20:
I believe you need to build it on your own machine to be able to get that sorted. I might be wrong.

For me, this built the `node-printer` or in npmjs: `printer` module correctly. But node-printer has problems too such as not being able to find the `.node` module built from the native. I changed the `./lib/printer.js` as such on line 6:
`binding_path = './node_modules/printer/lib/node_printer.node'` which came to the problem of this module being compiled under `NODE_MODULE_VERSION 57` but mine is 75.

Klemen1337, 2019-12-22:
@SimantoR so your problem was with the "printer" npm package and electron package rebuilding not "node-thermal-printer"?

SimantoR, 2019-12-22:
Yes. Had to downgrade electron to 5.0.6 and build printer module using node-gyp to finally get it to work.

Klemen1337, 2019-12-25:
@SimantoR Yeah rebuilding modules is a pain in the butt