Klemen1337/node-thermal-printer #123: How do i know interface of my printer

ertanatwork, 2019-11-01:
Hello, complete newbie here. I've got a Start TSP100. I'll connect it to a windows machine. How do I find the interface of the printer?

```
...
let printer = new ThermalPrinter({
  type: PrinterTypes.STAR,                                  // Printer type: 'star' or 'epson'
  interface: 'tcp://xxx.xxx.xxx.xxx', <<<--- How do i know its interface
...
```

ertankara, 2019-11-01:
Same here... Oh that's me! :) Please helpp :((

potamir, 2019-11-07:
actually it already explained in the documentation, but just use `interface: 'printer:your printer name' `or `interface:'printer:auto'` to use default printer in your windows. oh and dont forget to set the driver.
more or less like this:
`let printer = new ThermalPrinter({
      type: Types.EPSON,
      interface: "printer:thermal_printer",
      driver: require("printer"), //you must install printer module first
      options: {
        timeout: 0
      }
    });`

potamir, 2019-11-07:
actually it already explained in the documentation, but just use `interface: 'printer:your printer name' `or `interface:'printer:auto'` to use default printer in your windows. oh and dont forget to set the driver.
more or less like this:
`let printer = new ThermalPrinter({
      type: PrinterTypes.STAR,
      interface: "printer:thermal_printer", //or auto
      driver: require("printer"), //you must install printer module first
...
    });`

ertanatwork, 2019-11-07:
Thank you very much, ı'll look for more information into docs, I must have missed it.

closed 2019-11-07 by ertanatwork