Klemen1337/node-thermal-printer #125: Driver is undefined while calling `getPrinter`

ertankara, 2019-11-21:
Getting following error when I try to print a page

`"TypeError: Cannot read property 'getPrinter' of undefined"`

The error happens when executing
`/lib/interfaces/printer.js:31`

Because it doesn't meet the condition in the constructor to set driver.

I wonder what is supposed to pass as the second argument here and why it may be undefined?
`/lib/interfaces/index.js:17`

vadhack, 2019-11-26:
Please, i need help with this

vadhack, 2019-11-26:
This is the solution:

# file: node-thermal-printer/lib/interfaces/printer.js

class Printer extends Interface {
  constructor(printerName, moduleName) {
    super();
    this.name = printerName;
    if (moduleName && typeof moduleName === "object") {
      this.driver = moduleName;
    } else {
      this.driver = require(moduleName || (this.electron ? "electron-printer" : "printer"));
    }
  }

ertankara, 2019-11-26:
@vadhack I will try this, thanks for posting ❤️

Klemen1337, 2019-12-05:
@ertankara Can you share your code?

@vadhack The correct way would be to throw an error
```
if (moduleName && typeof moduleName === "object") {
  this.driver = moduleName;
} else {
  throw new Error("No driver set!");
}
```

ertankara, 2019-12-05:
@Klemen1337 

Using the code below in an Electron project.
```js

  const ThermalPrinter =  require("node-thermal-printer").printer;
  const PrinterTypes = require("node-thermal-printer").types;

  let printer = new ThermalPrinter({
        type: PrinterTypes.STAR,
        // I've also tried "printer:auto" both of them gives the same result
        // "TypeError: Cannot read property 'getPrinter' of undefined"
        interface: "printer:Star TSP100 Cutter (TSP143)",
        characterSet: "SLOVENIA",
        options: {
          timeout: 5000
        }
      });
      let isConnected = await printer.isPrinterConnected();
```

vadhack, 2019-12-05:
@Klemen1337  Yes, I realized,  thank you.

@ertankara the correct way is:
```javascript
      let printer = new ThermalPrinter({
        type: PrinterTypes.STAR,
        // I've also tried "printer:auto" both of them gives the same result
        // "TypeError: Cannot read property 'getPrinter' of undefined"
        interface: "printer:Star TSP100 Cutter (TSP143)",
        characterSet: "SLOVENIA",
        options: {
          timeout: 5000
        },
        driver: require(electron ? 'electron-printer' : 'printer'), // setting driver module
      });
      // Error is thrown when executing line below
      let isConnected = await printer.isPrinterConnected();
```

closed 2019-12-05 by ertankara