Klemen1337/node-thermal-printer #126: setTextSize doesn't work in Epson LX-350

vadhack, 2019-11-26:
In example: node-thermal-printer/examples/charsets.js

this is how print when call "printer.setTextSize(height, width)"

!wWow
!his is normal

add a some characters at start in buffer and the text size always is the same,

I used a printer: Epson LX-350

Can someone help me please?

vadhack, 2019-12-02:
Can someone help me?

Klemen1337, 2019-12-05:
@vadhack Not all printers have the same commands. Your printer might have a different set of commands.

Tested printers are:

- Star TSP700
- Rongta RP80US
- Rongta RP326-USE
- EPSON TM-T88V
- Posman BTP-R880NP (Type "epson")

vadhack, 2019-12-05:
Thanks, do you know where can i get the correct commands?

Klemen1337, 2019-12-06:
@vadhack I found them here [https://reference.epson-biz.com/modules/ref_escpos/index.php?content_id=2](https://reference.epson-biz.com/modules/ref_escpos/index.php?content_id=2)

closed 2019-12-06 by Klemen1337