Klemen1337/node-thermal-printer #127: Star BSC10 does not print Image correctly

CHOMNANP, 2019-11-28:
I'm trying to print the image with printer model Star BSC10. As a result, the image did not print properly as image. The output is look like a random binary string.

[image]

````
const ThermalPrinter = require("node-thermal-printer").printer;
const PrinterTypes = require("node-thermal-printer").types;

let printer = new ThermalPrinter({
    type: PrinterTypes.STAR,                                  // Printer type: 'star' or 'epson'
    interface: 'tcp://192.168.222.2',                       // Printer interface
    removeSpecialCharacters: false,                           // Removes special characters - default: false
    lineCharacter: "=",                                       // Set character for lines - default: "-"
    options: {                                                 // Additional options
        timeout: 5000                                           // Connection timeout (ms) [applicable only for network printers] - default: 3000
    }
});
await printer.printImage('./demo.png');  // Print PNG image
let execute = await printer.execute()

````

Look like an encoding issue. Anyone has similar experience?

CHOMNANP, 2019-11-28:
Tried to reproduce this in PHP and found the same issue.
Here's how they fix it.
https://github.com/mike42/escpos-php/issues/222

Klemen1337, 2019-12-05:
@CHOMNANP Not all printers have the same commands. Your printer might have a different print image command.

Tested printers are:
- Star TSP700
- Rongta RP80US
- Rongta RP326-USE
- EPSON TM-T88V
- Posman BTP-R880NP (Type "epson")

closed 2020-03-21 by stale[bot]