Klemen1337/node-thermal-printer #130: Epson Picture Printing Garbled

buddy-lin, 2019-12-25:
When the height of the picture is greater than or equal to 256 pixels, the picture will print garbled code. After testing, it has nothing to do with the amount of data.

Klemen1337, 2019-12-25:
@buddy-lin What printer are you using?

buddy-lin, 2019-12-25:
Thermal printer.However, pictures with a printing height less than 256 pixels can be printed normally

buddy-lin, 2019-12-25:
Thermal printer.
However, pictures with a printing height less than 256 pixels can be printed normally.

buddy-lin, 2019-12-25:
And the picture printing is in the middle. I want to use the command to change to left printing but it doesn't work

buddy-lin, 2019-12-30:
> @ buddy-lin您正在使用什么打印机？

Is the problem solved?

Klemen1337, 2020-01-07:
So just make image smaller than 256px?

buddy-lin, 2020-01-11:
Yes，Print as long as the height is less than 256 pixels.

closed 2020-04-30 by stale[bot]