Klemen1337/node-thermal-printer #133: printer.setPrinterDriver(MyCustomDriver) example object

odegraciajr, 2020-01-09:
May i know a sample driver object using `printer.setPrinterDriver(MyCustomDriver)`?

Im using xprinter thermal receipt printer

selenecodes, 2020-01-23:
You can pass it a node-printer driver or fork as far as I know. For node <12 try `require('printer')` and for node > 12 try `require('@thiagoelg/node-printer')`

selenecodes, 2020-01-23:
Also if you're using Electron with those you will need to rebuild.

For that you can use electron-rebuild using the following command
```
electron-rebuild -f -w @thiagoelg/node-printer
```

jevy-wangfei, 2020-04-20:
> May i know a sample driver object using `printer.setPrinterDriver(MyCustomDriver)`?
> 
> Im using xprinter thermal receipt printer

Hi, Does your xprinter work with this node-thermal-printer package?  
I see xprinter Compatible to ESC/POS command.

closed 2020-08-05 by stale[bot]