Klemen1337/node-thermal-printer #136: Remove electron-printer driver from readme

antoniomgatto, 2020-01-13:
Hi there !!

Thanks for this amazing module. I am developing a small app for my burger restaurant. So, I need to print the orders kkk

After some tests, I realized that is not necessary to use [electron printer](https://www.npmjs.com/package/electron-printer) anymore. [Node printer](https://www.npmjs.com/package/printer) was updated with electron support out of the box.

antoniomgatto, 2020-01-13:
pull request #137

Klemen1337, 2020-01-21:
@antoniomgatto You can still use electron printer without node printer. So why would you remove it?

antoniomgatto, 2020-01-25:
@Klemen1337 this project is only a fork. If you take a look into electron printer README there is a note 

"It's a temporary solution until binary package will be available for node 5.1.x and windows / linux platforms for node-printer on electron runtime."

closed 2020-05-12 by stale[bot]