Klemen1337/node-thermal-printer #138: Cyrillic characters

JamshedLatipov, 2020-01-17:
const p = new ThermalPrinter({
    type: PrinterTypes.EPSON,
    interface: 'printer:XP-58C (copy 2)',
    characterSet:  'SLOVENIA',
    removeSpecialCharacters: false,
    driver: nodePrinter,
     options: {
      driver: nodePrinter,
    },
  });
  p.setTextSize(0, 0);
  p.setTextQuadArea();
  p.print('фывапппываыфффффвавы');
  p.cut();
  await p.execute();

Muraveiko, 2020-01-19:
try characterSet:  'PC866_CYRILLIC2'

JamshedLatipov, 2020-01-20:
Thanks for response, but no luck PC866_CYRILLIC2, PC855_CYRILLIC and WPC1251_CYRILLIC.

Muraveiko, 2020-01-20:
I am sure that when printing a test page it will be written that the default encoding is GBK. Using the configuration utility from the printer manufacturer, set default for your language.
https://github.com/Klemen1337/node-thermal-printer/blob/master/lib/types/epson-config.js#L57
You need the code page number in the printer for your encoding. For example 17. If it does not match, you will have to write a config for your printer.

JamshedLatipov, 2020-01-21:
@Muraveiko when i runing [](https://github.com/Klemen1337/node-thermal-printer/blob/master/examples/charcode.js) and when i print any char(only one character) it's works perfectly but when i try to pint a string it's prints incorrect. Please help !)

Muraveiko, 2020-01-21:
GBK is a 2-byte encoding. Write a model and manufacturer. Show the self-test page. 
Are you sure that the necessary encoding is present in the printer firmware?
Cyrillic is present in GBK allsow.
FS . ( 28, dot) - command for china mode off (maybe another on you printer)

closed 2020-05-07 by stale[bot]