Klemen1337/node-thermal-printer #141: How to print HTML formatted text

MY020537, 2020-02-07:
Does this module support html template printing as well  anyhow ? i am able to print RAW text but not HTML formatted text.

Klemen1337, 2020-02-14:
@MY020537 this library creates buffer with commands specifically for epson and star printers.

closed 2020-02-14 by Klemen1337

WebScaffolder, 2022-02-24:
wha do you meant by create buffer, answer  yes or no or if possible then how?

nipoonp, 2023-01-10:
I made it work like this using the node-html-to-image library.
```

        const image = await nodeHtmlToImage({
            html: `<html><body><div style="font-size: 24px;">Check out what I just did! #cool</div></body></html>`,
            quality: 20,
        });

        console.log("xxx... image", image);

        await printer.printImageBuffer(image); // Print PNG image

```

fkoca, 2023-03-06:
Hi @nipoonp, your exemple didn't work for me. Nothing is printed. Do you have a special configuration?