Klemen1337/node-thermal-printer #142: Unhandled Rejection (TypeError): Net.connect is not a function

albertshah, 2020-02-21:
When I try to print, I get error : Unhandled Rejection (TypeError): Net.connect is not a function

I searched a lot and there was only one unanswered question like mine here: https://gitter.im/Klemen1337/node-thermal-printer?at=588fbd82c0f28dd862629227

I am trying to print from Reactjs project using Chrome on Mac, I tried it on Android device too but I get the same issue.

I installed all the dependencies like : net, write-file-queue, unorm, iconv-file too.

This is the print image code:
  testPrint = async () => {
    const ThermalPrinter = require("../../../node_modules/node-thermal-printer")
        .printer;
    const Types = require("../../../node_modules/node-thermal-printer").types;
    let printer = new ThermalPrinter({
        type: Types.EPSON,
        interface: "tcp://192.168.0.100:9100"
    });

    printer.alignCenter();
    printer.println("Hello world");

    try {
        let execute = printer.execute();
        console.error("Print done!");
    } catch (error) {
        console.log("Print failed:", error);
    }
 };

boris-savic, 2020-02-21:
Hi albertshah,

this is a **nodejs** library so it cannot be used in Reactjs or any other front-end library. 

The purpose of this library is to create a binary buffer of commands for thermal printers (Star and EPSON supported) and then stream those commands down to the printer. You can not do this from Chrome.

closed 2020-02-21 by Klemen1337

albertshah, 2020-02-21:
Thanks a lot Klemen for your prompt response.

1) I read in this post ( https://github.com/Klemen1337/node-thermal-printer/issues/74 ) that if we use Nwjs then we can use it with Reactjs too, I also installed NWjs but still the same error. Can you please clarify the solution.

2) As I a, new to node.js but found this package so useful, Is there any way that we can send the data from browser to a middleware by using intent or webview and then send the data to your package from that middleware or even directly?
Thanks in advance

boris-savic, 2020-02-23:
1. Nwjs (and also Electron) are projects that enable you to build native desktop apps for Windows/Linux/Mac with nodejs and whatever frontend framework you prefer. So you cannot just npm install it and expect your webapp that runs in chrome to work. I suggest you head over to official documentations of the respective projects and check out what they are and how to actually use them.

2. This question is pretty specific and unfortunately I cannot provide this level of support. You're not using the library correctly to start with - the prupose of this library is to construct the binary buffer of commands to send to the printer interface. You can only do that from Nodejs. I'd suggest you rethink your application and architecture.

kartiksolanki, 2020-08-07:
hi @albertshah, I am getting the same error inside my angular nodejs ionic web app. Did you find any solution?

Osuriel, 2021-05-29:
> * Nwjs (and also Electron) are projects that enable you to build native desktop apps for Windows/Linux/Mac with nodejs and whatever frontend framework you prefer. So you cannot just npm install it and expect your webapp that runs in chrome to work. I suggest you head over to official documentations of the respective projects and check out what they are and how to actually use them.
> * This question is pretty specific and unfortunately I cannot provide this level of support. You're not using the library correctly to start with - the prupose of this library is to construct the binary buffer of commands to send to the printer interface. You can only do that from Nodejs. I'd suggest you rethink your application and architecture.

Hey I know I might be asking for wayy to much but hopefully it doesn’t hurt to ask. I get your point about not being able to use this from the browser. But since you obviously got more experience than me or anyone I know I’ll ask you this. If I wanted to print from the browser to a thermal printer, is that possible at all? And we’re would you start researching? I’ve been researching the topic for over 6 months now and this library is the closest I’ve gotten to it so I figure maybe I’ll ask you for advise or a recommendation. Thank you in advance and if you’re too busy to answer this question I completely understand.

boris-savic, 2021-05-29:
> Hey I know I might be asking for wayy to much but hopefully it doesn’t hurt to ask. I get your point about not being able to use this from the browser. But since you obviously got more experience [...]

You could construct printable content in HTML and use standard Chrome/browser print function which then converts the page to PDF content, simply select your printer and you're good to go. 

This works on most printers if you have drivers installed

Elarabimo, 2023-05-16:
any update on this topic?