Klemen1337/node-thermal-printer #144: Node 13, invalid ELF header

autotel, 2020-03-04:
I don't know if I am forgetting something, or the examples found are bound to fail. I am using node 13 on Debian.

Steps I followed:
* create folder named "ttl-printer-test"
* open command line in that folder
* npm init -y
* npm install node-thermal-printer --save
* npm install electron-printer --save
* npm install electron --save
* npm install printer --save
* touch index.js
* open index.js to edit and added:

```
const ThermalPrinter = require("node-thermal-printer").printer;
const PrinterTypes = require("node-thermal-printer").types;
const electron = typeof process !== 'undefined' && process.versions && !!process.versions.electron;

(async ()=>{
    let printer = new ThermalPrinter({
    type: PrinterTypes.EPSON,
    interface: 'dev/ttyUSB0',
    driver: require(electron ? 'electron-printer' : 'printer')
    });

    printer.alignCenter();
    printer.println("Hello world");
    await printer.printImage('./assets/olaii-logo-black.png')
    printer.cut();
    
    try {
    let execute = printer.execute()
    console.error("Print done!");
    } catch (error) {
    console.log("Print failed:", error);
    }
})()
```

I get:

```
node index
(node:17703) UnhandledPromiseRejectionWarning: Error: /home/user/Desktop/ttl-printer-test/node_modules/printer/lib/node_printer.node: invalid ELF header
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:1201:18)
    at Module.load (internal/modules/cjs/loader.js:1000:32)
    at Function.Module._load (internal/modules/cjs/loader.js:899:14)
    at Module.require (internal/modules/cjs/loader.js:1040:19)
    at require (internal/modules/cjs/helpers.js:72:18)
    at Object.<anonymous> (/home/user/Desktop/ttl-printer-test/node_modules/printer/lib/printer.js:10:22)
    at Module._compile (internal/modules/cjs/loader.js:1151:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1171:10)
    at Module.load (internal/modules/cjs/loader.js:1000:32)
    at Function.Module._load (internal/modules/cjs/loader.js:899:14)
(node:17703) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:17703) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```

Klemen1337, 2020-05-28:
this is a problem with https://www.npmjs.com/package/printer package

closed 2020-05-28 by Klemen1337

thiagosanches, 2023-06-21:
Hey, you can try to use this fork: https://github.com/thiagoelg/node-printer.