Klemen1337/node-thermal-printer #148: Find printer in network

RWOverdijk, 2020-06-03:
Hey!

I love this module and it works really well. Thank you for this.

Currently I am configuring the IP address of the printer manually, but I would like to detect it on the network automatically. Is there a way I can do this? I can use "auto" for connected printers, but does this also work for printers on the network? Because it doesn't seem like it.  If there's a doc I didn't find/read I apologise. I did my best to find it before asking a question.

Thanks again, big fan 😄

Klemen1337, 2020-07-24:
Hi @RWOverdijk,
node-thermal-printer library is used to create buffer array for printers. Handling queues and discovering printers is a job for other libraries like https://www.npmjs.com/package/printer

closed 2020-07-24 by Klemen1337

RWOverdijk, 2020-07-24:
I understand. I tried that but it doesn't list printers on the network unfortunately. I could check every address in the network to see if I get a response I guess.

arcadia-dutchie, 2024-05-13:
@RWOverdijk Did you find an approach that worked for you?

RWOverdijk, 2024-05-13:
> @RWOverdijk Did you find an approach that worked for you?

I do not remember 😄 I solved it but it has been so long I can't remember how I did it sorry

arcadia-dutchie, 2024-05-13:
Darn! Thanks for getting back