Klemen1337/node-thermal-printer #149: Incomplete print

leonardovargasleffa, 2020-07-03:
I using module "node-thermal-printer" to create buffer and this for send data to print with node-printer.

[image]

But sometimes, part of the beginning of the text is not printed.
I save text in one file and save full text in file normally, but not printed full text.

Examples of buffer string

Print complete:
[5efd1b5d515e0996582c288a](https://github.com/thiagoelg/node-printer/files/4867561/5efd1b5d515e0996582c288a.txt)

Print incomplete:
[5efd152cc5804c81f94db8b3](https://github.com/thiagoelg/node-printer/files/4867562/5efd152cc5804c81f94db8b3.txt)
This cut the first 5 lines.

This not appear on second print.
[image]

samuelcseto, 2020-07-14:
Hello! I would have to see the whole code including the part where the print gets executed. Everything on your screenshot looks alright so the issue has to be somewhere else.

asaadbaw, 2020-10-25:
Hello

I had this issue too. I discovered that the printer object gets destroyed before the printing finish.

[image]

closed 2021-06-09 by stale[bot]

jortegamo, 2023-12-13:
Hi, I am facing the same issue but I am using the File interface. I mean, I am using the device path interface as Interface parameter. I have discovered that is due to the buffer size. I create all printing commands and then invoke to the execute method. The printer is printing only a certain data contained into the buffer. If I rerun the execute command then the printer prints part from the remaining data. 

I guess that all the buffer is send to the printer but it is not able to print the entire buffer at once. 

I am using an EPSON TM - t88iv. Maybe there ir any parameter into options to solve this problem? or maybe is an specific printer configuration? 

As a workaround I am using the .raw() method to print instantly and printer.clear() and several printer.execute() for each line. But I think that is not the optimal solution for this. 

Could me give me support about this? 

thanks a lot!