Klemen1337/node-thermal-printer #150: Is the library supported for TCS - TTP225 label printing model

rajjagani021, 2020-07-13:
I want to know is this library supports the TCS model. TTP. This printer prints the KOT information of the sticker. Can this library do that thing as well? 
[image]

samuelcseto, 2020-07-14:
Hello! This library supports EPSON and STAR communication protocols. The printer you want to use uses TSPL-EZ protocol which unfortunately isn't compatible. Please refer to this document if you want to interact with this printer. 
[TSPL-EZ Programming Manual](https://www.meto.com/metoprint/files/meto_print_programming_manual_tspl_ez_en.pdf)

Klemen1337, 2020-07-24:
@rajjagani021 label printers usually use different code and logic.

closed 2020-07-24 by Klemen1337