Klemen1337/node-thermal-printer #151: Electron React Boilerplate

binuchandrane, 2020-07-14:
I'm using Electron React Boilerplate to build my app. I'm getting the following error when I trying to load  ThermalPrinter  via
`const ThermalPrinter = require("node-thermal-printer").printer;`

`Uncaught TypeError: __webpack_require__(...) is not a function
    at Object../node_modules/node-thermal-printer/node_modules/iconv-lite/lib/index.js (index.js:144)
    at __webpack_require__ (bootstrap:789)
    at fn (bootstrap:100)
    at Object../node_modules/node-thermal-printer/lib/core.js (core.js:3)
    at __webpack_require__ (bootstrap:789)
    at fn (bootstrap:100)
    at Object../node_modules/node-thermal-printer/node-thermal-printer.js (node-thermal-printer.js:1)
    at __webpack_require__ (bootstrap:789)
    at fn (bootstrap:100)
    at Module../src/views/Dashboard/Dashboard.js (Dashboard.js:5)`
I tried to fix as the instructions added here https://github.com/Klemen1337/node-thermal-printer/issues/109 . But no luck. Please help

samuelcseto, 2020-07-14:
Hello! This is a known issue with the _iconv-lite_ package when used together with Electron. I have checked and it looks like it has already been fixed in release 0.6.0. Try to update to the latest version of this package and let us know if it helped. 😉
https://github.com/ashtuchkin/iconv-lite/commit/b106faaf15bb1bc66b20bdb81aa687415f54a7d4

binuchandrane, 2020-07-15:
It works. Thank you so much @samuelcseto

closed 2020-07-15 by binuchandrane

jonathantyar, 2020-12-12:
@binuchandrane can you please provide a clue how you can achieve using this package with Electron React Boilerplate, i got this issue to, but updating to the last version didn't solve the problem for me