Klemen1337/node-thermal-printer #153: Ionic Integration

kartiksolanki, 2020-08-07:
Hi, 
As ionic support nodejs, is it possible to integrate this plugin inside ionic pwa project?

closed 2020-11-24 by stale[bot]

HardikDG, 2021-02-06:
@kartiksolanki Does it worked with Ionic for you?

kartiksolanki, 2021-02-06:
@HardikDG : I have not tried this but I used [cordova-plugin-epos2](https://github.com/towerism/cordova-plugin-epson-epos2#readme) instead and it works wonderful. Also if you are using STAR printer then go to ionic API pages and they have a plugin for STAR as well.

davidquon, 2022-06-10:
I had this question as well trying to print from an Ionic PWA project.  Does anyone know if this is possible?

kartiksolanki, 2022-06-13:
@davidquon : For Ionic PWA project, I would suggest looking into [node-thermal-printer](https://www.npmjs.com/package/node-thermal-printer) which you can implement on your server side. If you are using Ionic for mobile platforms, use cordova-plugin-epos2 as I have implemented it and it works really well.